jerryscript-project/jerryscript #1535: Nominating Zidong Jiang (@jiangzidong) to become a JerryScript committer

zherczeg, 2017-01-19:
Hi,

I would like to nominate Zidong Jiang to become a JerryScript committer. He is a great developer and his work on typed arrays is great. I trust him to follow the process. His patches are:

#1507 [ES2015 profile]add TypedArray intrinsic object
#1479 Fix a bug in ArrayBuffer
#1467 add ES6 feature: ArrayBuffer
#1345 fix bug in vm call_stack_size calculation
#1277 fix bug: move jerry_make_api_unavailable into the end of jerry_cleanup
#1234 add target port: curie_bsp
#1217 fix bug of "deref bytecode twice" in jerry_exec_snapshot
#1210 Fix bug in jerry_string_to_char_buffer.
#836 remove the c++ syntax, struct::xxx
#806 refactor rcs_chunked_list and remove its c++ features
#720 fix a bug in linked_list memory allocation

zherczeg, 2017-01-27:
Congratulations Zidong! You are now a committer!

tilmannOSG, 2017-01-27:
Congratulations @jiangzidong!

closed 2017-01-27 by tilmannOSG

jiangzidong, 2017-01-27:
Thanks. @zherczeg @tilmannOSG Tomorrow is Chinese new year, happy new year and best wishes to JerryScript