jerryscript-project/jerryscript #1538: Board NRF52_DK: failure with make retarget.cpp:(.text.__wrap_main+0xa): undefined reference to `main'

OwenBrotherwood, 2017-01-22:
Using os ubuntu/xenial64,  the make BOARD=NRF52_DK failed with: 
```
Compile [100.0%]: launcher.cpp
Link: mbedos5
./BUILD/NRF52_DK/GCC_ARM/mbed-os/platform/retarget.o: In function `__wrap_main':
retarget.cpp:(.text.__wrap_main+0xa): undefined reference to `main'
collect2: error: ld returned 1 exit status
[ERROR] ./BUILD/NRF52_DK/GCC_ARM/mbed-os/platform/retarget.o: In function `__wrap_main':
retarget.cpp:(.text.__wrap_main+0xa): undefined reference to `main'
collect2: error: ld returned 1 exit status

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /home/ubuntu/jerryscript/targets/mbedos5/mbed-os/tools/make.py -t GCC_ARM -m NRF52_DK --source . --source ../../ --build ./BUILD/NRF52_DK/GCC_ARM -j0 -D CONFIG_MEM_HEAP_AREA_SIZE=(1024*16)" in "/home/ubuntu/jerryscript/targets/mbedos5"
---
Makefile:56: recipe for target 'all' failed
make: *** [all] Error 1

```

It would be interesting to know which ubuntu is usually sucessfull/accepted with compilation to remove as many unknowns as possible.

Basic preparation and make : have I forgotten something?
```
      sudo apt-get -y update \
      && sudo apt-get -y upgrade \
      && sudo apt-get -y install \
        cmake gcc-arm-none-eabi python-pip
      
      sudo pip install --upgrade pip \
      && sudo pip install -U mbed-cli \
      && git clone https://github.com/jerryscript-project/jerryscript.git \
      && cd jerryscript/targets/mbedos5 \
      && sudo pip install -r tools/requirements.txt \
      && make getlibs \
      && make BOARD=NRF52_DK

```

OwenBrotherwood, 2017-01-23:
Current status is:
- Build target jerry: Build suceeded!
- make BOARD=NRF52_DK: (new type of failure)

```
==> default: Compile [100.0%]: test_env.cpp
==> default: Link: mbedos5
==> default: ../../build/jerry-libc/CMakeFiles/jerry-libc.dir/target/posix/jerry-asm.S.o: file not recognized: File format not recognized
==> default: collect2: error: ld returned 1 exit status
==> default: [ERROR] ../../build/jerry-libc/CMakeFiles/jerry-libc.dir/target/posix/jerry-asm.S.o: file not recognized: File format not recognized
==> default: collect2: error: ld returned 1 exit status
==> default: [mbed] ERROR: "python" returned error code 1.
```

---

If one wishes, one can use the enclosed "Vagrantfile" with:
https://www.virtualbox.org/
https://www.vagrantup.com/

Install sw, place Vagrantfile in a directory, write "vagrant up" and let the vagrant run to see result.
The results are printed out on the command prompt.
"vagrant provision" can be run to see results if alterations in the sw pulled.

```
Vagrant.configure("2") do |config|
  config.vm.box = "ubuntu/xenial64"
  config.vm.synced_folder "../data", "/vagrant_data"
  config.vm.provision "shell", inline: <<-SHELL
      echo "=== Provision start"

      echo "== Ensure dist is up to date"
      apt-get -y update \
      && apt-get -y upgrade \
      && apt-get -y install \
        cmake gcc-arm-none-eabi python-pip
      
      echo "== pip up to date and install mbed-cli, with update"
      pip install --upgrade pip \
      && pip install -U mbed-cli \      

      echo "== git jerryscript"
      echo "= Will fail with vagrant provision on a provisioned box: ignore if this is the case"
      git clone https://github.com/jerryscript-project/jerryscript.git 

      echo "== target jerry"
      cd jerryscript \
      git pull \
      && python tools/build.py

      echo "== target mbedos5"     
      cd targets/mbedos5 \
      && pip install -r tools/requirements.txt

      echo "= board NRF52_DK"
      make getlibs \
      && make BOARD=NRF52_DK

      echo "== clean up"
      chown -R ubuntu ~ubuntu
      echo "=== Provision stop"
  SHELL
end
```

janjongboom, 2017-01-23:
@OwenBrotherwood Can you try with this program? https://github.com/armmbed/mbed-js-example

Build instructions change over time, so we wrapped them in a gulp module... Should update the blog post.

janjongboom, 2017-01-23:
Also, which version of GCC are you using? Try with GNU ARM Embedded Toolchain 4.9.3 (not 5).

OwenBrotherwood, 2017-01-23:
```
ubuntu@ubuntu-xenial:~$ arm-none-eabi-gcc-4.9.3 --version
arm-none-eabi-gcc-4.9.3 (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

OwenBrotherwood, 2017-01-23:
```
Link: mbedos5
Elf2Bin: mbedos5
+-----------------------+--------+-------+-------+
| Module                |  .text | .data |  .bss |
+-----------------------+--------+-------+-------+
| Fill                  |    332 |     4 |    21 |
| Misc                  | 303183 |  3644 | 18992 |
| drivers               |   3280 |     0 |    92 |
| events                |     52 |     0 |     0 |
| events/equeue         |   1432 |     0 |    61 |
| hal                   |    550 |     0 |    16 |
| platform              |   1727 |     4 |   292 |
| rtos                  |    802 |     4 |     4 |
| rtos/rtx              |   6791 |    20 |  4278 |
| targets/TARGET_NORDIC |  13836 |    20 |  1108 |
| Subtotals             | 331985 |  3696 | 24864 |
+-----------------------+--------+-------+-------+
Allocated Heap: 22688 bytes
Allocated Stack: 2048 bytes
Total Static RAM memory (data + bss): 28560 bytes
Total RAM memory (data + bss + heap + stack): 53296 bytes
Total Flash memory (text + data + misc): 335681 bytes

Object file test_env.o is not unique! It could be made from: ./mbed-os/features/frameworks/greentea-client/source/test_env.cpp /home/ubuntu/mbed-js-example/build/jerryscript/targets/mbedos5/mbed-os/features/unsupported/tests/mbed/env/test_env.cpp
Image: ../../../out/NRF52_DK/mbedos5.hex

```

OwenBrotherwood, 2017-01-23:
I then tried in the  mbed-js-example/build/jerryscript/targets/mbedos5 with a  make BOARD=NRF52_DK that gives a bad result:
```

Link: mbedos5
./BUILD/NRF52_DK/GCC_ARM/mbed-os/platform/retarget.o: In function `__wrap_main':
retarget.cpp:(.text.__wrap_main+0xa): undefined reference to `main'
collect2: error: ld returned 1 exit status
[ERROR] ./BUILD/NRF52_DK/GCC_ARM/mbed-os/platform/retarget.o: In function `__wrap_main':
retarget.cpp:(.text.__wrap_main+0xa): undefined reference to `main'
collect2: error: ld returned 1 exit status

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /vagrant_data/mbed-js-example/build/jerryscript/targets/mbedos5/mbed-os/tools/make.py -t GCC_ARM -m NRF52_DK --source . --source ../../ --build ./BUILD/NRF52_DK/GCC_ARM -j0 -D CONFIG_MEM_HEAP_AREA_SIZE=(1024*16)" in "/vagrant_data/mbed-js-example/build/jerryscript/targets/mbedos5"
---
Makefile:56: recipe for target 'all' failed
make: *** [all] Error 1

```

OwenBrotherwood, 2017-01-23:
Changed NO_JS=1 to NO_JS=0
in /mbed-js-example/build/Makefile

(just banging the keyboard to change things and see a result: I will see if there is anything from the Makefile I can use in ubuntu build)

```
ubuntu@ubuntu-xenial:/vagrant_data/mbed-js-example/build$ make BOARD=NRF52_DK
cd jerryscript/targets/mbedos5 && pip install -r tools/requirements.txt && rm -f source/js_encoded.cpp && rm -f source/pins.cpp && rm -f source/main.cpp && make BOARD=NRF52_DK EXTRA_SRC="../../../source" EXTERN_BUILD_DIR=../../../out/NRF52_DK NO_JS=0
Requirement already satisfied: pycparser in /usr/local/lib/python2.7/dist-packages (from -r tools/requirements.txt (line 1))
Requirement already satisfied: simpleeval in /usr/local/lib/python2.7/dist-packages (from -r tools/requirements.txt (line 2))
Requirement already satisfied: pycparserext in /usr/local/lib/python2.7/dist-packages (from -r tools/requirements.txt (line 3))
Requirement already satisfied: ply>=3.4 in /usr/local/lib/python2.7/dist-packages (from pycparserext->-r tools/requirements.txt (line 3))
make[1]: Entering directory '/vagrant_data/mbed-js-example/build/jerryscript/targets/mbedos5'
python ../tools/js2c.py --ignore pins.js
python tools/generate_pins.py NRF52_DK
mbed target NRF52_DK
[mbed] NRF52_DK now set as default target in program "mbedos5"
mbed compile -j0 --source . --source ../../ --source ../../../source --build ../../../out/NRF52_DK -D "CONFIG_MEM_HEAP_AREA_SIZE=(1024*16)" -t GCC_ARM
Building project mbedos5 (NRF52_DK, GCC_ARM)
Scan: .
Scan: FEATURE_UVISOR
Scan: FEATURE_LWIP
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_BLE
Scan: FEATURE_STORAGE
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan:
Scan: source
Scan: mbed
Scan: env
Compile [100.0%]: pins.cpp
[Warning] pins.cpp@205,5: this decimal constant is unsigned only in ISO C90
Link: mbedos5
Elf2Bin: mbedos5
+-----------------------+--------+-------+-------+
| Module                |  .text | .data |  .bss |
+-----------------------+--------+-------+-------+
| Fill                  |    332 |     4 |    21 |
| Misc                  | 303183 |  3644 | 18992 |
| drivers               |   3280 |     0 |    92 |
| events                |     52 |     0 |     0 |
| events/equeue         |   1432 |     0 |    61 |
| hal                   |    550 |     0 |    16 |
| platform              |   1727 |     4 |   292 |
| rtos                  |    802 |     4 |     4 |
| rtos/rtx              |   6791 |    20 |  4278 |
| targets/TARGET_NORDIC |  13836 |    20 |  1108 |
| Subtotals             | 331985 |  3696 | 24864 |
+-----------------------+--------+-------+-------+
Allocated Heap: 22688 bytes
Allocated Stack: 2048 bytes
Total Static RAM memory (data + bss): 28560 bytes
Total RAM memory (data + bss + heap + stack): 53296 bytes
Total Flash memory (text + data + misc): 335681 bytes

Object file test_env.o is not unique! It could be made from: ./mbed-os/features/frameworks/greentea-client/source/test_env.cpp /vagrant_data/mbed-js-example/build/jerryscript/targets/mbedos5/mbed-os/features/unsupported/tests/mbed/env/test_env.cpp
Image: ../../../out/NRF52_DK/mbedos5.hex
make[1]: Leaving directory '/vagrant_data/mbed-js-example/build/jerryscript/targets/mbedos5'

```

OwenBrotherwood, 2017-01-23:
I am wondering if I have misinterpreted the intention of jerryscript on a micro-controller.
Do I need a javascript job for jerryscript compared to espruino: onboardjavascript?

janjongboom, 2017-01-24:
@OwenBrotherwood I'm not completely sure what you're trying to do right now. It looks like you're missing `main.cpp` which should be generated during the build... Probably a path issue when running the makefile.

However, the paths should be set if you build with `gulp --target=NRF52_DK`. Does that produce a proper build?

This worked fine for me on Ubuntu 15.10, with node.js v6.3.0, gcc 4.9.3 and Python 2.7:

```
janjon01@ubuntu-janjon01:~/$ npm install -g gulp
janjon01@ubuntu-janjon01:~/$ pip install mbed-cli
janjon01@ubuntu-janjon01:~/$ git clone git@github.com:armmbed/mbed-js-example.git
janjon01@ubuntu-janjon01:~/$ cd mbed-js-example
janjon01@ubuntu-janjon01:~/mbed-js-example$ gulp --target=NRF52_DK
```

What do you mean with:

> Do I need a javascript job for jerryscript compared to espruino: onboardjavascript?

This runs a JS interpreter on the microcontroller, so no need for anything else.

OwenBrotherwood, 2017-01-26:
> What do you mean with:

> - Do I need a javascript job for jerryscript compared to espruino: onboardjavascript?
> This runs a JS interpreter on the microcontroller, so no need for anything else.

Great: this is also what I thought, but I needed to make sure.

I go through the build again and see if I can find where my/the problem is

OwenBrotherwood, 2017-01-26:
https://github.com/ARMmbed/mbed-js-example just seems to be on the surface, make an led blink and does not indicate an on board javascript interpreter that can be interacted with like the espruino where a cmd prompt is available on the microprocessor
https://www.youtube.com/watch?v=T3YbwAtgrcg

janjongboom, 2017-01-26:
@OwenBrotherwood See http://github.com/janjongboom/mbed-js-repl-example

OwenBrotherwood, 2017-01-26:
Looks like what I was wanting: I will try it
As such, my problems with compiling jerryscript have been in relation to a non-gulp method.
Gulp methods seem to work so I expect to close the issue soon with a suggestion to have something in this repo about gulp methods.

zherczeg, 2017-03-04:
Was there any success fixing it?

OwenBrotherwood, 2017-03-05:
@zherczeg I did not really return to jerryscript: do you have a problem?

zherczeg, 2017-03-05:
Others might be interested how you fixed it if they have the same problem.

OwenBrotherwood, 2017-03-05:
WIP This comments gets editted
git clone --recursive https://github.com/jerryscript-project/jerryscript.git

zherczeg, 2017-03-31:
Is this issue resolved?

janjongboom, 2017-03-31:
Yes.

closed 2017-03-31 by janjongboom

OwenBrotherwood, 2017-03-31:
Sorry I am not able to use more time on jerryscript: I find it interesting but micro:bit cannot and the other boards are Espruino purchases
In all case "on board javascript" for zx80 on steroids: the kids of today have some cool stuff to play and learn on.