jerryscript-project/jerryscript #1547: Assertion '!ecma_number_is_nan (time)' failed  in ecma_date_year_from_time

renatahodovan, 2017-01-30:
###### Jerry version:

```
Checked revision: 8b5d645
Build: debug.linux
```
###### OS:

```
Ubuntu 16.04, x86_64
```
###### Test case:

```javascript

try { new (this.Date)(this.Number .MAX_VALUE, this.Number .MAX_VALUE) .$ () } catch($){}
```

###### Backtrace:

```
ICE: Assertion '!ecma_number_is_nan (time)' failed at JERRY_PATH/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.c(ecma_date_year_from_time):114.
Error: ERR_FAILED_INTERNAL_ASSERTION
Aborted
bt
#0  0x000000000046d5c0 in syscall_2 () at JERRY_PATH/jerry-libc/target/posix/jerry-asm.S:59
#1  0x00000000004027be in raise (sig=6) at JERRY_PATH/jerry-libc/target/posix/jerry-libc-target.c:91
#2  0x0000000000402790 in abort () at JERRY_PATH/jerry-libc/target/posix/jerry-libc-target.c:77
#3  0x0000000000411450 in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at JERRY_PATH/targets/default/jerry-port-default-fatal.c:53
#4  0x000000000042cf04 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at JERRY_PATH/jerry-core/jrt/jrt-fatals.c:61
#5  0x000000000042cf56 in jerry_assert_fail (assertion=0x46f85a "!ecma_number_is_nan (time)", file=0x46f7f0 "JERRY_PATH/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.c", function=0x487890 <__func__.3294.lto_priv.88> "ecma_date_year_from_time", line=114) at JERRY_PATH/jerry-core/jrt/jrt-fatals.c:85
#6  0x000000000040d34c in ecma_date_year_from_time (time=-nan(0x8000000000000)) at JERRY_PATH/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.c:114
#7  0x000000000040e158 in ecma_date_make_day (year=1.7976931348623157e+308, month=1.7976931348623157e+308, date=1) at JERRY_PATH/jerry-core/ecma/builtin-objects/ecma-builtin-helpers-date.c:514
#8  0x00000000004453be in ecma_date_construct_helper (args=0x7fffffffccac, args_len=2) at JERRY_PATH/jerry-core/ecma/builtin-objects/ecma-builtin-date.c:149
#9  0x000000000044612c in ecma_builtin_date_dispatch_construct (arguments_list_p=0x7fffffffccac, arguments_list_len=2) at JERRY_PATH/jerry-core/ecma/builtin-objects/ecma-builtin-date.c:530
#10 0x0000000000454f5a in ecma_builtin_dispatch_construct (obj_p=0x691290 <jerry_global_heap+152>, arguments_list_p=0x7fffffffccac, arguments_list_len=2) at JERRY_PATH/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:153
#11 0x0000000000438d88 in ecma_op_function_construct (func_obj_p=0x691290 <jerry_global_heap+152>, arguments_list_p=0x7fffffffccac, arguments_list_len=2) at JERRY_PATH/jerry-core/ecma/operations/ecma-function-object.c:712
#12 0x000000000041dd96 in opfunc_construct.lto_priv.137 (frame_ctx_p=0x7fffffffcd00) at JERRY_PATH/jerry-core/vm/vm.c:443
#13 0x0000000000415735 in vm_execute (frame_ctx_p=0x7fffffffcd00, arg_p=0x0, arg_list_len=0) at JERRY_PATH/jerry-core/vm/vm.c:2600
#14 0x000000000041595e in vm_run (bytecode_header_p=0x692e60 <jerry_global_heap+7272>, this_binding_value=27, lex_env_p=0x691220 <jerry_global_heap+40>, is_eval_code=false, arg_list_p=0x0, arg_list_len=0) at JERRY_PATH/jerry-core/vm/vm.c:2664
#15 0x000000000041d8ec in vm_run_global (bytecode_p=0x692e60 <jerry_global_heap+7272>) at JERRY_PATH/jerry-core/vm/vm.c:211
#16 0x00000000004696e9 in jerry_run (func_val=139) at JERRY_PATH/jerry-core/jerry.c:340
#17 0x000000000046d0bb in main (argc=3, argv=0x7fffffffd348) at JERRY_PATH/jerry-main/main-unix.c:711
```

zherczeg, 2017-02-05:
I wasn't aware that 

    (Infinity % 1) == NaN

I would say it is 0, but it seems not.

pfalcon, 2017-02-05:
Were this and other recent bugreports of this form produced https://github.com/renatahodovan/fuzzinator ?  If so, might be interesting to add some footer to a report linking to it as a kind of advertisement ;-)  (I'm not sure what "ease the reporting of issues in bug trackers (e.g., Bugzilla or GitHub)" in its description means and whether it produces output to copy-paste or goes as far as posting directly, but would work for both.)

pfalcon, 2017-02-05:
> I wasn't aware that (Infinity % 1) == NaN

Here's apparently why:
~~~
0.1 % 1 == 0.1
~~~

So, infinity modulo 1 is all the real numbers in range [0, 1) at the same time, which is NaN.

zherczeg, 2017-02-06:
Hm, there is one more interesting computation:

    Infinity - 1 == Infinity
    Infinity - Infinity == NaN

This actually explains the modulo behaviour:

    modulo = a - trunc(a/n) where a = Infinity, n = 1.0

renatahodovan, 2017-02-06:
@pfalcon: Yeah, those bugs come from Fuzzinator. I was also thinking about marking fuzzers bugs somehow, like I do in bugzilla with a common metabug. Unfortunately, GH doesn't have such a concept and I cannot assign labels either. However, footers can work actually - if nobody has objection agains them (?). (And yeah, Fuzzinator posts directly to GH using to a user-configured issue template).

PR #1620 (2017-02-27): NaN must not be passed to date getter functions.
Fixes #1547
commit 3de72af7 (2017-02-27): NaN must not be passed to date getter functions. (#1620)

Fixes #1547

JerryScript-DCO-1.0-Signed-off-by: Zoltan Herczeg zherczeg.u-szeged@partner.samsung.com
closed 2017-02-27 by zherczeg