jerryscript-project/jerryscript #1549: Assertion 'lit_is_cesu8_string_valid (string_p, string_size)' failed  in ecma_new_ecma_string_from_utf8

renatahodovan, 2017-01-31:
###### Jerry version:

```
Checked revision: 7b59c80
Build: debug.linux
```
###### OS:

```
Ubuntu 16.04, x86_64
```
###### Test case:

```javascript

function $() {

$ = {
$:$, $: ($ ($, ($ * ($[(($ + '(Cqc=V'.$ ())) & $] + $))))
}
}
```

###### Backtrace:

```
ICE: Assertion 'lit_is_cesu8_string_valid (string_p, string_size)' failed at jerryscript/jerry-core/ecma/base/ecma-helpers-string.c(ecma_new_ecma_string_from_utf8):154.
Error: ERR_FAILED_INTERNAL_ASSERTION
Aborted
bt
#0  0x000000000046daba in syscall_2 () at jerryscript/jerry-libc/target/posix/jerry-asm.S:59
#1  0x0000000000402760 in raise (sig=6) at jerryscript/jerry-libc/target/posix/jerry-libc-target.c:91
#2  0x0000000000402732 in abort () at jerryscript/jerry-libc/target/posix/jerry-libc-target.c:77
#3  0x00000000004113c6 in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/targets/default/jerry-port-default-fatal.c:53
#4  0x000000000042ce6f in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.c:61
#5  0x000000000042cec1 in jerry_assert_fail (assertion=0x47c788 "lit_is_cesu8_string_valid (string_p, string_size)", file=0x47c700 "jerryscript/jerry-core/ecma/base/ecma-helpers-string.c", function=0x480e30 <__func__.3095.lto_priv.558> "ecma_new_ecma_string_from_utf8", line=154) at jerryscript/jerry-core/jrt/jrt-fatals.c:85
#6  0x000000000045ae6c in ecma_new_ecma_string_from_utf8 (string_p=0x712be0 <buffer+2496> "B!\262E\264' + '(Cq\324c=V\314'.indexOf(strvar5))) & 255] + ary[((((true instanceof ((typeof EvalError == 'function') ? EvalError : Object)) >= 0 ? (true instanceof ((typeof EvalError == 'function') ? EvalError :"..., string_size=5) at jerryscript/jerry-core/ecma/base/ecma-helpers-string.c:154
#7  0x0000000000413790 in ecma_find_or_create_literal_string (chars_p=0x712be0 <buffer+2496> "B!\262E\264' + '(Cq\324c=V\314'.indexOf(strvar5))) & 255] + ary[((((true instanceof ((typeof EvalError == 'function') ? EvalError : Object)) >= 0 ? (true instanceof ((typeof EvalError == 'function') ? EvalError :"..., size=5) at jerryscript/jerry-core/ecma/base/ecma-literal-storage.c:72
#8  0x000000000042a4c6 in parser_post_processing (context_p=0x7fffffffcbc0) at jerryscript/jerry-core/parser/js/js-parser.c:1715
#9  0x000000000042b3c8 in parser_parse_function (context_p=0x7fffffffcbc0, status_flags=6) at jerryscript/jerry-core/parser/js/js-parser.c:2155
#10 0x0000000000425380 in lexer_construct_function_object (context_p=0x7fffffffcbc0, extra_status_flags=6) at jerryscript/jerry-core/parser/js/js-lexer.c:1639
#11 0x000000000041e347 in parser_parse_function_statement (context_p=0x7fffffffcbc0) at jerryscript/jerry-core/parser/js/js-parser-statm.c:396
#12 0x0000000000420d32 in parser_parse_statements (context_p=0x7fffffffcbc0) at jerryscript/jerry-core/parser/js/js-parser-statm.c:1688
#13 0x000000000042a99c in parser_parse_source.lto_priv.202 (source_p=0x712220 <buffer> "//", '-' <repeats 103 times>, "\n// Copyright (C) Microsoft. All rights reserved.\n// Licensed under the MIT license. See LICENS"..., size=2915, strict_mode=0, error_location_p=0x7fffffffcd70) at jerryscript/jerry-core/parser/js/js-parser.c:1874
#14 0x00000000004227b1 in parser_parse_script (source_p=0x712220 <buffer> "//", '-' <repeats 103 times>, "\n// Copyright (C) Microsoft. All rights reserved.\n// Licensed under the MIT license. See LICENS"..., size=2915, is_strict=false, bytecode_data_p=0x7fffffffcdb0) at jerryscript/jerry-core/parser/js/js-parser.c:2252
#15 0x00000000004699fe in jerry_parse (source_p=0x712220 <buffer> "//", '-' <repeats 103 times>, "\n// Copyright (C) Microsoft. All rights reserved.\n// Licensed under the MIT license. See LICENS"..., source_size=2915, is_strict=false) at jerryscript/jerry-core/jerry.c:262
#16 0x000000000046d580 in main (argc=3, argv=0x7fffffffd2e8) at jerryscript/jerry-main/main-unix.c:706
```

zherczeg, 2017-02-10:
This can be simplified to the quoted string:

27 D4 63 3D 56 CC 27

The quoted string is not a valid UTF-8 string. D4 63 is not a valid UTF-8 character.

In JerryScript the user must verify the source code (it comes from a trusted source or checked before passed to jerry parse), so this report is invalid.

zherczeg, 2017-02-10:
To be more precise: when an invalid UTF-8 source code is passed to jerry_parse, the behaviour is undefined. Crashing is acceptable in this case.

PR #1576 (2017-02-16): Add new input validator API functions
Fixes #1549

JerryScript-DCO-1.0-Signed-off-by: László Langó llango.u-szeged@partner.samsung.com
commit 799726aa (2017-02-16): Add new input validator API functions (#1576)

Fixes #1549

JerryScript-DCO-1.0-Signed-off-by: László Langó llango.u-szeged@partner.samsung.com
closed 2017-02-16 by dbatyai