jerryscript-project/jerryscript #1550: Assertion 'JERRY_CONTEXT (jmem_heap_allocated_size) == 0' failed  in jmem_heap_finalize

renatahodovan, 2017-01-31:
###### Jerry version:

```
Checked revision: 7b59c80
Build: debug.linux
```
###### OS:

```
Ubuntu 16.04, x86_64
```
###### Test case:

```javascript
do break; while $
```

###### Backtrace:

```
Script Error: SyntaxError
ICE: Assertion 'JERRY_CONTEXT (jmem_heap_allocated_size) == 0' failed at jerryscript/jerry-core/jmem/jmem-heap.c(jmem_heap_finalize):180.
Error: ERR_FAILED_INTERNAL_ASSERTION
Aborted
bt
#0  0x000000000046daba in syscall_2 () at jerryscript/jerry-libc/target/posix/jerry-asm.S:59
#1  0x0000000000402760 in raise (sig=6) at jerryscript/jerry-libc/target/posix/jerry-libc-target.c:91
#2  0x0000000000402732 in abort () at jerryscript/jerry-libc/target/posix/jerry-libc-target.c:77
#3  0x00000000004113c6 in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/targets/default/jerry-port-default-fatal.c:53
#4  0x000000000042ce6f in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.c:61
#5  0x000000000042cec1 in jerry_assert_fail (assertion=0x4757b0 "JERRY_CONTEXT (jmem_heap_allocated_size) == 0", file=0x475720 "jerryscript/jerry-core/jmem/jmem-heap.c", function=0x4848e0 <__func__.2985.lto_priv.265> "jmem_heap_finalize", line=180) at jerryscript/jerry-core/jrt/jrt-fatals.c:85
#6  0x000000000042c26e in jmem_heap_finalize () at jerryscript/jerry-core/jmem/jmem-heap.c:180
#7  0x000000000042cb7b in jmem_finalize () at jerryscript/jerry-core/jmem/jmem-allocator.c:59
#8  0x000000000046d93d in jerry_cleanup () at jerryscript/jerry-core/jerry.c:168
#9  0x000000000046d803 in main (argc=3, argv=0x7fffffffd2e8) at jerryscript/jerry-main/main-unix.c:806
```

LaszloLango, 2017-01-31:
This issue is quite interesting. Every parse error has a memory leak. Something has been broken in `main-unix.c`

LaszloLango, 2017-01-31:
My bad. The bug is related to wrong break statements only.

PR #1558 (2017-02-03): Stack data should be freed for do-while statements after the expression is parsed.
Fixes #1550.
commit a43186db (2017-02-03): Stack data should be freed for do-while statements after the expression is parsed. (#1558)

Fixes #1550.

JerryScript-DCO-1.0-Signed-off-by: Zoltan Herczeg zherczeg.u-szeged@partner.samsung.com
closed 2017-02-03 by zherczeg