jerryscript-project/jerryscript #1562: [Question about PR] typedarray prototype methods: one big patch or seperate ones

jiangzidong, 2017-02-06:
Hi 
I am working on the typedarray prototype methods, such as `every` `filter` `forEach` `map` `some` `indexOf` `lastIndexOf` `reduce` `reduceRight` ....

Should I sumbit a whole big patch or classify those methods into several patches?

Thanks
Zidong

LaszloLango, 2017-02-06:
IMHO, it would be nice to do it in separate patches. One PR for one routine. It would make the reviews much faster and you can work on the next PR while we are reviewing the others.

zherczeg, 2017-02-06:
As for me simple ones could go together, but big ones should be separate patches.

jiangzidong, 2017-02-06:
Thanks! I will separate them into groups which contains the similar routines, e.g. foreach, every, some will be in one patch

closed 2017-02-07 by jiangzidong