jerryscript-project/jerryscript #1564: Is there any plan to support for ESP32?

limjunsung, 2017-02-06:
I want to build jerryscript for ESP32(http://esp32.net/), but I found no solution or tips. Is there any plan to support for ESP32?

LaszloLango, 2017-02-06:
Someone reported on our IRC channel that JerryScript works on ESP32.

slaff, 2017-02-09:
@limjunsung Compiles nicely on ESP8266 and since the toolchain for ESP32 is similar to ESP8266 I would expect that it should compile also there without issues. Cannot confirm though because I do not have ESP32 to test.

chalkers, 2017-02-10:
@slaff It now works on the ESP8266? I thought there was a slow down here - #1375

slaff, 2017-02-10:
It works as a library since, if I am not mistaken end of spring, last year. You just need to compile the stock JerryScript for esp8266 using the appropriate CMake settings and then you can use the produced static library (search for *.a file). In my case I using mostly jerry-core, since the math functions are already provided from the microc library that I am using.

tilmannOSG, 2017-02-10:
@limjunsung As @LaszloLango mentioned already, we had someone visiting the IRC channel who mentioned that he got JerryScript running on the ESP32 in two hours (basically porting it himself).

young-mu, 2017-02-21:
It works fine for ESP32, but you need to use ENABLE_ALL_IN_ONE feature.

akosthekiss, 2017-07-27:
Closing due to inactivity. (Conclusion seems to be that there is no actual support/example for ESP32 in the repo and core developers don't have it as main priority. However, broader community has had success porting JerryScript easily based on the in-repo ESP8266 port/example -- but that work has not been contributed back.)

closed 2017-07-27 by akosthekiss