jerryscript-project/jerryscript #1595: ENABLE_ALL_IN_ONE feature bug

young-mu, 2017-02-20:
I need to use the ENABLE_ALL_IN_ONE feature to integrate jerryscript to my project. But the generated build/jerry-all-in.c does NOT include files under jerry-core/debugger/, which makes the feature unusable. So please fix it ASAP. Thanks.

LaszloLango, 2017-02-20:
@young-mu, how did you try to build? The following command worked for me as expected:
`tools/build.py --all-in-one=on --debug --clean`

young-mu, 2017-02-20:
Exactly the same. You can reproduce this bug as follows.

1. Save the [example code](https://github.com/jerryscript-project/jerryscript/blob/master/docs/03.API-EXAMPLE.md#step-1-execute-javascript-from-your-application) as test.c
```c
#include <string.h>
#include "jerryscript.h"

int
main (int argc, char *argv[])
{
  const jerry_char_t script[] = "print ('Hello, World!');";
  size_t script_size = strlen ((const char *) script);

  bool ret_value = jerry_run_simple (script, script_size, JERRY_INIT_EMPTY);

  return (ret_value ? 0 : 1);
}
```
2. Compile jerryscript
```shell
tools/build.py --all-in-one=on --debug --clean
```

3. Compile test.c and jerry-all-in.c into a.out
```shell
gcc ~/Github/jerryscript/build/jerry-all-in.c test.c \                                      
    -I ~/Github/jerryscript/jerry-core \
    -I ~/Github/jerryscript/jerry-core/jrt \
    -I ~/Github/jerryscript/jerry-core/lit \
    -I ~/Github/jerryscript/jerry-core/jmem \
    -I ~/Github/jerryscript/jerry-core/jcontext \
    -I ~/Github/jerryscript/jerry-core/ecma/builtin-objects \
    -I ~/Github/jerryscript/jerry-core/ecma/builtin-objects/typedarray \
    -I ~/Github/jerryscript/jerry-core/ecma/base \
    -I ~/Github/jerryscript/jerry-core/debugger \
    -I ~/Github/jerryscript/jerry-core/parser/regexp \
    -I ~/Github/jerryscript/jerry-core/vm \
    -I ~/Github/jerryscript/jerry-core/parser/js \
    -I ~/Github/jerryscript/jerry-core/ecma/operations \
    ~/Github/jerryscript/build/lib/libjerry-libm.a \
    -DJMEM_STATS \
    -DJERRY_ENABLE_SNAPSHOT_SAVE \
    -DJERRY_ENABLE_SNAPSHOT_EXEC
```

4. Execute a.out
```shell
./a.out
```

You will see nothing output.

---

**But, If you checkout the v1.0 branch of jerryscript. And change the jerryscript.h to jerry-api.h (actually they are the same) and recompile, rerun the executable. `Hello, World!` will be shown.**

zherczeg, 2017-02-20:
You don't have a parser since you didn't define JERRY_JS_PARSER, and JS parsing returns with error. The parser is not a mandatory feature since snapshot support is enough to run pre-compiled javascript.

young-mu, 2017-02-21:
After adding **JERRY_JS_PARSER** build option, it can work for me. Thanks for your comment.

closed 2017-02-21 by young-mu