jerryscript-project/jerryscript #1605: Assertion 'ecma_get_value_type_field (value) == ECMA_TYPE_OBJECT' failed in ecma_get_object_from_value

bzsolt, 2017-02-21:
###### Environment: ######
``` 
Ubuntu 16.04
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.4)
Jerry: 92eaea2
```

###### Build command: ######
`tools/build.py --debug --profile=es2015-subset --compile-flag=-m32 --system-allocator=on --jerry-libc=off`

###### Test case: ######
```javascript
var a = new Int8Array();
```

###### Error: ######
```
Assertion 'ecma_get_value_type_field (value) == ECMA_TYPE_OBJECT' failed at jerryscript/jerry-core/ecma/base/ecma-helpers-value.c(ecma_get_object_from_value):605
```

###### Backtrace: ######
```
#0  0xf7ffdbe9 in __kernel_vsyscall ()
#1  0x080f2847 in raise ()
#2  0x080b9f87 in abort ()
#3  0x080649d2 in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION)
    at jerryscript/targets/default/jerry-port-default-fatal.c:53
#4  0x0807a3d4 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION)
    at jerryscript/jerry-core/jrt/jrt-fatals.c:61
#5  0x0807a407 in jerry_assert_fail (assertion=0x8140030 "ecma_get_value_type_field (value) == ECMA_TYPE_OBJECT", 
    file=0x813fdec "jerryscript/jerry-core/ecma/base/ecma-helpers-value.c", 
    function=0x8129334 <__func__.3617.lto_priv.629> "ecma_get_object_from_value", line=609)
    at jerryscript/jerry-core/jrt/jrt-fatals.c:85
#6  0x080a1703 in ecma_get_object_from_value (value=0)
    at jerryscript/jerry-core/ecma/base/ecma-helpers-value.c:609
#7  0x08080d32 in ecma_typedarray_get_arraybuffer (typedarray_p=0x837fbb8)
    at jerryscript/jerry-core/ecma/operations/ecma-typedarray-object.c:462
#8  0x080a702e in ecma_gc_mark.lto_priv.647 (object_p=0x837fbb8)
    at jerryscript/jerry-core/ecma/base/ecma-gc.c:284
#9  0x080a0a49 in ecma_gc_run (severity=JMEM_FREE_UNUSED_MEMORY_SEVERITY_LOW)
    at jerryscript/jerry-core/ecma/base/ecma-gc.c:654
#10 0x080a0c15 in ecma_free_unused_memory (severity=JMEM_FREE_UNUSED_MEMORY_SEVERITY_LOW)
    at jerryscript/jerry-core/ecma/base/ecma-gc.c:742
#11 0x08079fe2 in jmem_run_free_unused_memory_callbacks (severity=JMEM_FREE_UNUSED_MEMORY_SEVERITY_LOW)
    at jerryscript/jerry-core/jmem/jmem-allocator.c:148
#12 0x0807a269 in jmem_heap_gc_and_alloc_block (size=24, ret_null_on_error=false)
    at jerryscript/jerry-core/jmem/jmem-heap.c:357
#13 0x0807a32c in jmem_heap_alloc_block (size=24) at jerryscript/jerry-core/jmem/jmem-heap.c:406
#14 0x0809cfac in ecma_alloc_extended_object (size=24) at jerryscript/jerry-core/ecma/base/ecma-alloc.c:99
#15 0x080ab27c in ecma_create_object (prototype_object_p=0x837fa10, ext_object_size=24, type=ECMA_OBJECT_TYPE_GENERAL)
    at jerryscript/jerry-core/ecma/base/ecma-helpers.c:103
#16 0x0809dde1 in ecma_builtin_init_object (obj_builtin_id=ECMA_BUILTIN_ID_ARRAYBUFFER_PROTOTYPE, prototype_obj_p=0x837fa10, 
    obj_type=ECMA_OBJECT_TYPE_GENERAL, is_extensible=true)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c:163
#17 0x0809f462 in ecma_instantiate_builtin (id=ECMA_BUILTIN_ID_ARRAYBUFFER_PROTOTYPE)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:304
#18 0x0809dcab in ecma_builtin_get (builtin_id=ECMA_BUILTIN_ID_ARRAYBUFFER_PROTOTYPE)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c:111
#19 0x08079be7 in ecma_arraybuffer_new_object (length=0)
    at jerryscript/jerry-core/ecma/operations/ecma-arraybuffer-object.c:89
#20 0x08080533 in ecma_typedarray_create_object_with_length (array_length=0, proto_p=0x837fb98, element_size_shift=0 '\000', 
    class_id=LIT_MAGIC_STRING_INT8_ARRAY_UL)
    at jerryscript/jerry-core/ecma/operations/ecma-typedarray-object.c:225
#21 0x08080eff in ecma_op_create_typedarray (arguments_list_p=0xffffc954, arguments_list_len=0, proto_p=0x837fb98, 
    element_size_shift=0 '\000', class_id=LIT_MAGIC_STRING_INT8_ARRAY_UL)
    at jerryscript/jerry-core/ecma/operations/ecma-typedarray-object.c:555
#22 0x08082fca in ecma_builtin_int8array_dispatch_construct (arguments_list_p=0xffffc954, arguments_list_len=0)
    at jerryscript/jerry-core/ecma/builtin-objects/typedarray/ecma-builtin-int8array.c:69
```

bzsolt, 2017-02-21:
@jiangzidong could you check it please?

jiangzidong, 2017-02-24:
I will check

closed 2017-02-27 by LaszloLango