jerryscript-project/jerryscript #1613: Assertion 'ecma_builtin_is (obj_p, ECMA_BUILTIN_ID_GLOBAL)' failed in ecma_object_get_class_name

renatahodovan, 2017-02-23:
###### Jerry version:

```
Checked revision: 62a581f
Build: debug.linux
```
###### OS:

```
Ubuntu 16.04.2 LTS
```
###### Test case:

```javascript
new Int16Array(Float32Array.prototype)
```
###### Build command:
```
tools/build.py --clean --debug --profile=es2015-subset --compile-flag=-m32 --system-allocator=on --jerry-libc=off
```

###### Backtrace:

```
ICE: Assertion 'ecma_builtin_is (obj_p, ECMA_BUILTIN_ID_GLOBAL)' failed at jerryscript/jerry-core/ecma/operations/ecma-objects.c(ecma_object_get_class_name):1762.
Error: ERR_FAILED_INTERNAL_ASSERTION
bt
#0  0xf7ffdbe9 in __kernel_vsyscall ()
#1  0x080f21d7 in raise ()
#2  0x080b9d17 in abort ()
#3  0x08057732 in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/targets/default/jerry-port-default-fatal.c:53
#4  0x0807076b in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.c:61
#5  0x0807079e in jerry_assert_fail (assertion=0x8139e20 "ecma_builtin_is (obj_p, ECMA_BUILTIN_ID_GLOBAL)", file=0x8139780 "jerryscript/jerry-core/ecma/operations/ecma-objects.c", function=0x812c5c4 <__func__.4429.lto_priv.425> "ecma_object_get_class_name", line=1762) at jerryscript/jerry-core/jrt/jrt-fatals.c:85
#6  0x0807b98d in ecma_object_get_class_name (obj_p=0x837e958) at jerryscript/jerry-core/ecma/operations/ecma-objects.c:1762
#7  0x08076237 in ecma_is_arraybuffer (target=137881947) at jerryscript/jerry-core/ecma/operations/ecma-arraybuffer-object.c:146
#8  0x080751f6 in ecma_op_create_typedarray (arguments_list_p=0xffffc064, arguments_list_len=1, proto_p=0x837e998, element_size_shift=1 '\001', class_id=LIT_MAGIC_STRING_INT16_ARRAY_UL) at jerryscript/jerry-core/ecma/operations/ecma-typedarray-object.c:594
#9  0x080804bc in ecma_builtin_int16array_dispatch_construct (arguments_list_p=0xffffc064, arguments_list_len=1) at jerryscript/jerry-core/ecma/builtin-objects/typedarray/ecma-builtin-int16array.c:69
#10 0x0809af08 in ecma_builtin_dispatch_construct (obj_p=0x837e7d8, arguments_list_p=0xffffc064, arguments_list_len=1) at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:373
#11 0x0807c943 in ecma_op_function_construct (func_obj_p=0x837e7d8, arguments_list_p=0xffffc064, arguments_list_len=1) at jerryscript/jerry-core/ecma/operations/ecma-function-object.c:712
#12 0x08062909 in opfunc_construct.lto_priv.245 (frame_ctx_p=0xffffc084) at jerryscript/jerry-core/vm/vm.c:443
#13 0x0805ae1d in vm_execute (frame_ctx_p=0xffffc084, arg_p=0x0, arg_list_len=0) at jerryscript/jerry-core/vm/vm.c:2656
#14 0x0805af9c in vm_run (bytecode_header_p=0x837e8f0, this_binding_value=137881147, lex_env_p=0x837e6e8, is_eval_code=false, arg_list_p=0x0, arg_list_len=0) at jerryscript/jerry-core/vm/vm.c:2730
#15 0x0806246b in vm_run_global (bytecode_p=0x837e8f0) at jerryscript/jerry-core/vm/vm.c:211
#16 0x080b2263 in jerry_run (func_val=137881499) at jerryscript/jerry-core/jerry.c:385
#17 0x080b1beb in main (argc=3, argv=0xffffc4d4) at jerryscript/jerry-main/main-unix.c:726
```

<sup>Found by  [Fuzzinator](http://fuzzinator.readthedocs.io/)</sup>

LaszloLango, 2017-02-23:
cc: @jiangzidong

jiangzidong, 2017-02-24:
I will check it

closed 2017-02-27 by LaszloLango