jerryscript-project/jerryscript #1630: tests/jerry/parser-oom.js fail

jiangzidong, 2017-03-01:
Jerry version
```
checked version e993ee6c
```

OS
```
Ubuntu 14.04 64bit
```

build cmd
```
./tools/build.py --clean --debug --compile-flag=-m32 --system-allocator=on --jerry-libc=off
```

Test case
```
./build/bin/jerry tests/jerry/parser-oom.js

Script Error: assertion failed
```

If I remove the build flag: --system-allocator=on, then the case will pass

zherczeg, 2017-03-01:
As the name suggests, this is an out-of-memory test specifically tailored for 512K max RAM. When you have more RAM, if does not fail.

jiangzidong, 2017-03-01:
Sorry I didn't fully understand. Do you mean that my RAM is less than 512K when I set `--system-allocator=on`, so it failed?

zherczeg, 2017-03-01:
The opposite. Your RAM is more than 512K so it is failed :)

zherczeg, 2017-03-01:
With 16 bit compressed pointers the RAM is limited to 512K and the test is designed that in mind. If you have more than 512K, the test fails.

zherczeg, 2017-03-01:
Sorry, I see I accidentally wrote "if does not fail." I wanted "it fails".

LaszloLango, 2017-03-01:
The heap limit option won't work, when you are using the system allocator. That test is intended to check the engine won't crash when it runs out of memory, but throws an exception and terminates normally.

LaszloLango, 2017-03-01:
`tests/jerry/parser-oom.js` should be skipped when testing the `--system-allocator=on` option. I think @bzsolt has been working on it.

jiangzidong, 2017-03-01:
Thanks. And I just realized I made a stupid mistake. I thought the `assert` will throw a error and catched by catch block... That's the reason I cannot understand why large RAM will cause error...Haha
```
try
{
 ...
  assert (false);
}
catch(e)
{
...
}
```

closed 2017-03-03 by jiangzidong