jerryscript-project/jerryscript #1635: Runtime error messages contain no source information

tadziopazur, 2017-03-02:
I happen to run my JS code with jerry engine.
There are issues with the code (my bad), but all I get in the end is:

uncaughtException: ReferenceError: Cannot resolve reference.

That's it.

How do I get some information regarding error location?
File and line would be the bare minimum.
Column and identifier would help even more.
Getting some context would be extremely helpful, as I am likely to immediately see what is wrong.

LaszloLango, 2017-03-02:
Try to use the debugger: https://github.com/jerryscript-project/jerryscript/blob/master/docs/07.DEBUGGER.md

My first guess is a typo and you are trying to access a property of a null object. You might also interested in https://github.com/jerryscript-project/jerryscript/pull/1577

Currently line and column information is provided for parse errors only (if error messages are enabled).

zherczeg, 2017-03-02:
We are planning to add a feature to the debugger which shows where the exception is thrown.

polaroi8d, 2017-03-28:
Hey @tadziopazur, now we're working on it. Recently opened a PR which has this feature, you can check it in #1693

zherczeg, 2017-03-31:
The PR is merged, now you can use the debugger to capture exceptions.

closed 2017-03-31 by zherczeg