jerryscript-project/jerryscript #1647: JerryScript uses unaligned word access when dealing with bytecode

pfalcon, 2017-03-07:
This issue was discovered and debugged by @MaureenHelm. She got a Zephyr.js build with a sample application, which is converted by Zephyr.js to bytecode snapshot, faulting on FRDM-K64F (ARM Cortex-M4). The same sample worked for me, so it took some time to debug. Maureen found that it boils down to these lines: https://github.com/jerryscript-project/jerryscript/blob/master/jerry-core/ecma/base/ecma-literal-storage.c#L481 . As can be seen, the input pointer is uint8_t*, but gets casted to uint32_t* and dereferenced. This is not safe conversion for many architectures, where unaligned access to words, etc. is not supported.

What's interested is that Cortex-M3/4 actually support unaligned access (but not Cortex-M0 for example), but it works only with single-word (or half-word) loads. And he we had a compiler optimizing 2 consecutive loads to multiple load instruction, which still doesn't support unaligned access:

~~~
uint32_t string_count = ((uint32_t *) buffer_p)[0];                     
uint32_t number_count = ((uint32_t *) buffer_p)[1];
    6a1a:       e890 0410       ldmia.w r0, {r4, sl}
~~~

So, the above code is definitely a bug in JerryScript, and there possibly more, so more code should be reviewed.

We'd appreciate a comment from the maintainers if there's already a way to resolve this (i.e. a function/macro to do byte-by-byte loads and then combination of individual bytes). We'd be looking into this over next days as time permits otherwise.

pfalcon, 2017-03-07:
@grgustaf, @poussa : FYI

zherczeg, 2017-03-07:
If buffer_p is aligned to 4 bytes, that must be aligned to 4 bytes as well. Unless the data is corrupted.

MaureenHelm, 2017-03-08:
buffer_p wasn't aligned to 4 bytes

zherczeg, 2017-03-10:
The current design expects that. Could you copy the data into an aligned space?

poussa, 2017-03-10:
@zherczeg Why can't you change the design, fix the issue and expect less from the application?

tilmannOSG, 2017-03-10:
@pfalcon @MaureenHelm Thank you very much for tracking down this issue, I fully agree that this is a bug and  this should be fixed in JerryScript rather than expecting the user to pass in a 32-bit aligned buffer.

zherczeg, 2017-03-10:
I would like to understand the use case first. Both the generator and the loader expects 4 byte aligned buffers now, so the easiest fix would be to change the API to expect an uint32_t buffer (size could be changed to number-of-words instead of bytes). If we would like to support byte aligned buffers we would need to add a lot of memcpy which reduces performance. The length of the  buffer produced by the generator is always divisible by 8 bytes.

pfalcon, 2017-03-10:
> The current design expects that.

I'm currently at the conference, so couldn't spend enough time to find an answer to the following questions myself (as I usually do): 1) Is the implementation guarantees that if beginning of a bytecode array is 32-bit aligned, then that particular function will be called with 32-bit aligned pointer? 2) Is the requirement that input bytecode should be 32-bit aligned is documented, and is it documented enough?

In that function alone, nothing says that it expects 32-bit aligned pointer, it takes uint8_t*, no param annotation, no comment, nothing. And it's not safe to take uint8_t*, cast it to something else and dereference.

All in all, we can call this only the problem of Zephyr.js and fix it there, or see if something can be improved on JrS side too. (And it took some head-scratching and time to debug it, I wouldn't wish any other project/hw arch to go thru that again and again).

zherczeg, 2017-03-10:
If you would be ok with changing the buffer_p to `uint32_t*` that would be the best.

tilmannOSG, 2017-03-10:
@pfalcon As @zherczeg said it's probably not worth the trouble to make the API work with an 8-bit aligned pointer for the buffer. The straightforward fix seems to be to just make it explicit that the function is consuming a 32-bit aligned pointer (obviously requiring a small change to the API).  
From what I can see, none of those alignment requirements are currently documented so this is definitely something that needs to be improved.  
We are currently in the process of auditing the source code for other problematic pointer casts that violate alignment constraints. Hopefully all of those issues will be fixed soon.

zherczeg, 2017-03-17:
I hope the landed patch fixes this issue for you.

closed 2017-03-17 by zherczeg