jerryscript-project/jerryscript #1649: old-style-definition warnings

emmanuelsearch, 2017-03-08:
Compiling Jerryscript (+ RIOT) on my Linux VM, I get the below old-style-definition warnings (by default treated as errors, though they can be bypassed by tweaking the CFLAGS).

```
/home/vagrant/RIOT/examples/javascript/bin/pkg/iotlab-m3/jerryscript/jerry-core/jmem/jmem-allocator.c:40:1: warning: old-style function definition [-Wold-style-definition]
 jmem_finalize ()
 ^

/home/vagrant/RIOT/examples/javascript/bin/pkg/iotlab-m3/jerryscript/jerry-core/jmem/jmem-poolman.c: In function 'jmem_pools_collect_empty':
/home/vagrant/RIOT/examples/javascript/bin/pkg/iotlab-m3/jerryscript/jerry-core/jmem/jmem-poolman.c:194:1: warning: old-style function definition [-Wold-style-definition]
 jmem_pools_collect_empty ()
 ^

/home/vagrant/RIOT/examples/javascript/bin/pkg/iotlab-m3/jerryscript/jerry-core/parser/regexp/re-compiler.c: In function 're_cache_gc_run':
/home/vagrant/RIOT/examples/javascript/bin/pkg/iotlab-m3/jerryscript/jerry-core/parser/regexp/re-compiler.c:491:1: warning: old-style function definition [-Wold-style-definition]
 re_cache_gc_run ()
 ^

/home/vagrant/RIOT/examples/javascript/bin/pkg/iotlab-m3/jerryscript/targets/default/jerry-port-default-date.c: In function 'jerry_port_get_current_time':
/home/vagrant/RIOT/examples/javascript/bin/pkg/iotlab-m3/jerryscript/targets/default/jerry-port-default-date.c:53:8: warning: old-style function definition [-Wold-style-definition]
 double jerry_port_get_current_time ()
        ^

/home/vagrant/RIOT/examples/javascript/bin/pkg/iotlab-m3/jerryscript/targets/default/jerry-port-default-fatal.c: In function 'jerry_port_default_is_abort_on_fail':
/home/vagrant/RIOT/examples/javascript/bin/pkg/iotlab-m3/jerryscript/targets/default/jerry-port-default-fatal.c:39:6: warning: old-style function definition [-Wold-style-definition]
 bool jerry_port_default_is_abort_on_fail ()
      ^
```

akosthekiss, 2017-03-08:
@emmanuelsearch could you please add some details? What is the version of JerryScript you are using (git hash)? Which compiler are you using and which version? How do you compile JerryScript (what is your build command and/or what are your cmake arguments)?

emmanuelsearch, 2017-03-09:
Sorry for not giving more details in the first place:
Commit: e993ee6cdc955218a534cb397da9e95eeaf8b5c8
Obtained using: make jerry-core (invoked in [here](https://github.com/jerryscript-project/jerryscript/blob/master/targets/riot-stm32f4/Makefile.riot)).
Compiler: arm-none-eabi-gcc 4.9.3

PR #1651 (2017-03-09): Fix old-style function definitions.
Function definitions with no parameters should always use the void keyword to allow the compiler to catch invalid calls of those functions at compile time.

Fixes #1649.

JerryScript-DCO-1.0-Signed-off-by: Tilmann Scheller t.scheller@samsung.com
commit f8dd54ab (2017-03-09): Fix old-style function definitions and enable warning. (#1651)

Function definitions with no parameters should always use the void keyword to allow the compiler to catch invalid calls of those functions at compile time.

Enable -Wold-style-definition to catch this early in the future.

Fixes #1649.

JerryScript-DCO-1.0-Signed-off-by: Tilmann Scheller t.scheller@samsung.com
closed 2017-03-09 by tilmannOSG