jerryscript-project/jerryscript #1665: Discussion: ES6 import / System.import?

martijnthe, 2017-03-17:
The first thing that comes to mind thinking about `import` is probably to include another piece of .js source, but that's not the main reason why I'm interested in this feature:

I think having this ES6 feature would help a lot in creating an ecosystem of modules/API bindings for JerryScript. Having a standard way to define, register, instantiate (and destroy) a module, paves the way for reuse. It would make it much easier to take someone else's module and plug it into your project.

I’m still familiarizing myself with this part of the spec, but is there anything in ES6 import / System.import that would make it impossible to add this to JerryScript?

martijnthe, 2017-03-27:
Closing this. After more thinking and discussing with some other people, I think it's possible to come up with a different solution that doesn't depend on `import`.

closed 2017-03-27 by martijnthe

martijnthe, 2018-09-06:
@wille-io this ended up driving the efforts behind the module functionality in JerryScript, see http://jerryscript.net/ext-reference-module/

With a bit of code it allows you to implement a `require()`-like function. [One of the test suites] has a rudimentary version of that.

In our project we ended up doing that + using a transpiler (i.e. Babel or the TypeScript compiler) to translate `import` into `require()` calls.

I guess it's a stepping stone towards `import`.

[One of the test suites]: https://github.com/jerryscript-project/jerryscript/blob/master/tests/unit-ext/module/jerry-module-test.c

galpeter, 2018-09-06:
If I read the standard correctly:

The way the `import` works it is very different from the way `require` works. require does dynamic module loading, that is the module is processed when the require function is called during runtime. However, the `import` (and `export`) is done before executing any code in the given js and it is a static loader.