jerryscript-project/jerryscript #1670: Assertion '!ecma_is_value_empty (ret_value)' failed in ecma_builtin_dispatch_call

renatahodovan, 2017-03-21:
###### Jerry version:

```
Checked revision: 4420401
Build command: ./tools/build.py --clean --debug --profile=es2015-subset --compile-flag=-m32 --system-allocator=on --jerry-libc=off --error-messages=on
```
###### OS:

```
Ubuntu 16.04.2 LTS
```
###### Test case:

```javascript
Int16Array.from();
```

The same assertion check fails when calling the `of` method or using any other typed array variants (Int8Array, Uint8ClampedArray, Uint32Array, etc.).

###### Backtrace:

```
ICE: Assertion '!ecma_is_value_empty (ret_value)' failed at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c(ecma_builtin_dispatch_call):883.
Error: ERR_FAILED_INTERNAL_ASSERTION
bt
#0  0xf7fd8be9 in __kernel_vsyscall ()
#1  0xf7898ea9 in raise () from /lib/i386-linux-gnu/libc.so.6
#2  0xf789a407 in abort () from /lib/i386-linux-gnu/libc.so.6
#3  0x0806add9 in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION)
    at jerryscript/targets/default/jerry-port-default-fatal.c:53
#4  0x080a3f56 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.c:61
#5  0x080a3f8e in jerry_assert_fail (assertion=0x8148160 "!ecma_is_value_empty (ret_value)", 
    file=0x8147f60 "jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c", 
    function=0x8115c00 <__func__.5013.lto_priv.627> "ecma_builtin_dispatch_call", line=883)
    at jerryscript/jerry-core/jrt/jrt-fatals.c:85
#6  0x080e532a in ecma_builtin_dispatch_call (obj_p=0xf6500a00, this_arg_value=4132440627, arguments_list_p=0xffffc04c, arguments_list_len=0)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c:883
#7  0x080b1293 in ecma_op_function_call (func_obj_p=0xf6500a00, this_arg_value=4132440627, arguments_list_p=0xffffc04c, arguments_list_len=0)
    at jerryscript/jerry-core/ecma/operations/ecma-function-object.c:461
#8  0x0807d0de in opfunc_call.lto_priv.237 (frame_ctx_p=0xffffc0b0) at jerryscript/jerry-core/vm/vm.c:411
#9  0x0806837a in vm_execute (frame_ctx_p=0xffffc0b0, arg_p=0x0, arg_list_len=0) at jerryscript/jerry-core/vm/vm.c:2674
#10 0x080688ff in vm_run (bytecode_header_p=0xf6500ac0, this_binding_value=4132441011, lex_env_p=0xf61007f0, is_eval_code=false, arg_list_p=0x0, 
    arg_list_len=0) at jerryscript/jerry-core/vm/vm.c:2753
#11 0x0807c787 in vm_run_global (bytecode_p=0xf6500ac0) at jerryscript/jerry-core/vm/vm.c:231
#12 0x0810de04 in jerry_run (func_val=4132440723) at jerryscript/jerry-core/jerry.c:387
#13 0x0810d1f3 in main (argc=3, argv=0xffffc5b4) at jerryscript/jerry-main/main-unix.c:731
```

<sup>Found by  [Fuzzinator](http://fuzzinator.readthedocs.io/)</sup>

akosthekiss, 2017-03-22:
cc @jiangzidong

jiangzidong, 2017-03-25:
fixed in #1679

closed 2017-03-25 by jiangzidong