jerryscript-project/jerryscript #1673: Nominating Jan Jongboom (@janjongboom) to become a JerryScript committer and Embed port reviewer

zherczeg, 2017-03-22:
Hi,

Jan did an excellent work on porting and maintaining JerryScript on the ARM Mbed OS so I would like to nominate him to become a committer for JerryScript and reviewer for the Mbed port.

His patches are:

#1531 target: mbedos5: Change to debug profiles
#1522 target: mbedos5: Support for building on Windows
#1518 target: mbedos5: Sort magic strings
#1512 target: mbedos5: mbed CLI refuses to build if no .mbed file present
#1494 target: mbedos5: Update generate_pins.py
#1485 target: mbedos5: Define JERRY_JS_PARSER macro
#1468 target: mbedos5: Update to mbed OS 5.2.3
#1458 target: mbedos5: Allow jerry_port_console and jerry_port_log to be overriden by user code
#1457 target: mbedos5: Add return value to setTimeout/setInterval and implement clearTimeout/clearInterval
#1455 target: mbedos5: Add PwmOut and AnalogIn interfaces
#1427 target: mbedos5: Add carriage return in jerry_port_console

Any seconds?

martijnthe, 2017-03-22:
+1 (don't know if my vote has any meaning ;))

akosthekiss, 2017-03-28:
Congrats to the successful nomination, @janjongboom!

closed 2017-03-28 by akosthekiss

janjongboom, 2017-03-28:
[image]