jerryscript-project/jerryscript #1681: [Proposal] Add `type` argument to native handle API

martijnthe, 2017-03-23:
So far, all projects I've looked at that use JerryScript's [native handle APIs], fail to implement proper validation on obtained native handle values. Zephyr.js, IoT.js, mbed.js, etc. are all affected by this problem.

Let me explain the problem by using an example from the mbedos5 port. The port defines a number of bindings that define classes of JavaScript objects which, on the native side, are backed by different C++ classes (in case of a C project, they would be backed by different types of structs).

For example, there is a [`AnalogIn` C++ class] for which a [`AnalogIn` JavaScript constructor and method bindings] are defined. Below a snippet for the `read()` method. There, `jerry_get_object_native_handle()` is used to get a pointer to the `AnalogIn` C++ instance that is (presumably) associated with the JS `AnalogIn` object on which `read()` is called:

```
// AnalogIn.prototype.read(num):
DECLARE_CLASS_FUNCTION(AnalogIn, read) {
    CHECK_ARGUMENT_COUNT(AnalogIn, read, (args_count == 0));

    // Extract native AnalogIn pointer
    uintptr_t ptr_val;
    jerry_get_object_native_handle(this_obj, &ptr_val);

    AnalogIn* native_ptr = reinterpret_cast<AnalogIn*>(ptr_val);

    float result = native_ptr->read();
    return jerry_create_number(result);
}
```

The problem arises when `this` gets bound to a JS object with no native handle, or even worse, a native handle containing a pointer to a *different* C++ class instance / C struct type. For example:

```
// d's native handle is a pointer to C++ instance of class `DigitalOut`:
var d = new DigitalOut(1);

// i's native handle is a pointer to C++ instance of class `AnalogIn`:
var i = new AnalogIn(2);

// Whoops.. the read() binding will interpret the native handle a pointer to an `AnalogIn` instance, while it's actually a `DigitalOut`... BAD!
i.read.call(d);
```

Note this is not only applies to getting native handles for `this`, but also for arguments to a function. Also note that checking `this.constructor` isn't bullet proof, because it can be changed from the JS code.

## TL;DR

So, in short, one would *always* have to add type checks to make sure that the native handle is actually of the expected type, before casting and dereferencing the handle's value.

Because one would always have to deal when using the native handle API and because apparently the issue is overlooked by many dependent projects, I would argue that this concern should be addressed by JerryScript's API, by adding a type field to the API:

```
// Proposed, new API:
bool jerry_get_object_typed_native_handle (const jerry_value_t obj_val,
                                           uintptr_t *out_handle_p,
                                           uintptr_t *out_type_p);

void jerry_set_object_typed_native_handle (const jerry_value_t obj_val,
                                           uintptr_t handle_p,
                                           uintptr_t type,
                                           jerry_object_free_callback_t freecb_p);
// Old API:

__attribute__((deprecated))
bool jerry_get_object_native_handle (const jerry_value_t obj_val,
                                     uintptr_t *out_handle_p);
__attribute__((deprecated))
void jerry_set_object_native_handle (const jerry_value_t obj_val,
                                     uintptr_t handle_p,
                                     jerry_object_free_callback_t freecb_p);
```

## Implementation thoughts

- Instead of adding yet another "external pointer" internal property to store the type, perhaps the handle, type and free callback pointer can be combined into a single struct to reduce the memory footprint. (What's the rationale behind the handle value and free callback being stored as separate internal properties in today's code?)
- A pointer and type can probably be compressed into a `uintptr_t` on small embedded systems, but the "compression" would be dependent on the memory layout of the target, may require some .ld script mangling, etc. So, not very convenient. If the additional memory footprint is an issue, a solution may be to let the proposed API call into `jerry_port_...` functions to compress/decompress the type and pointer into a `uintptr_t` (and provide an example / readme on how to implement them).

[native handle APIs]: https://github.com/jerryscript-project/jerryscript/blob/4420401cf5c7810a8c5c9404e6921d88d30c16ad/jerry-core/jerryscript.h#L326
[`AnalogIn` C++ class]: https://docs.mbed.com/docs/mbed-os-api-reference/en/5.1/APIs/io/AnalogIn/
[`AnalogIn` JavaScript constructor and method bindings]: https://github.com/jerryscript-project/jerryscript/blob/4420401cf5c7810a8c5c9404e6921d88d30c16ad/targets/mbedos5/jerryscript-mbed/jerryscript-mbed-drivers/source/AnalogIn-js.cpp#L32

jiangzidong, 2017-03-23:
Martijn and I discussed about this proposal before. We met the problem in our project and think it deserves a discussion in community. 

Besides reminding/guiding  jerry-based developers to write the robust binding-layer, another reason is that it is not very convenient to implement own type-validation-code in binding-layer. e.g. one possible way is whenever call `jerry_set_object_native_handle`, malloc a wrapper struct, whose `handler` member store the native handle and `type` member store the type info, and let the wrapper struct be the js object's native handler.
It will be much easier for us if jerry_object_t itself provides a space to store the native type info.

If we concern about the memory footprint, we can add macro like `JERRY_SAFE_MODE` and let developer to choose based on their device capacity. If not defined `JERRY_SAFE_MODE`, then the `typeinfo` in 
```
void jerry_set_object_typed_native_handle (const jerry_value_t obj_val,
                                           uintptr_t handle_p,
                                           uintptr_t type,
                                           jerry_object_free_callback_t freecb_p);
```
will be unused.

zherczeg, 2017-03-23:
This is a reasonable idea. AS for me type and free_cb is quite close in nature. What about using the free_cb to store the type?

jiangzidong, 2017-03-23:
but free_cb will be called automatically when the js_obj is GCed, and it is also a very useful field. Do you mean to compress `free_cb` and `type_info` into a uintptr_t ?

zherczeg, 2017-03-23:
For me handle_p is something unique for all objects while type and freecb_p are shared. Allocating a a structure with 3 members, and two usually shared across many objects is kind of waste. Or it should be acceptable?

martijnthe, 2017-03-24:
@zherczeg that's a clever idea. Of course, that solution assumes that there is a 1:1 relation between `free_cb` and the type of object/struct. Although you can write the binding code to make sure that there is this 1:1 relation, I can also imagine writing a `free_cb` that is shared among different types of objects/structs.

If there is a way to enforce this 1:1 relationship at compile-time in a simple way (i.e. by using `_Static_assert()`), I'd be fine with using `free_cb` as a proxy for the object/struct type.

But I don't how to express such an assertion / whether it is possible in the first place.

martijnthe, 2017-03-24:
I just had another idea that does not increase the memory footprint.

What if:

```
// --- jerryscript.h

typedef struct {
  uintptr_t type;
  jerry_object_free_callback_t free_cb;
} jerry_native_handle_info_t;

#define JERRY_NATIVE_HANDLE_INFO_FOR_CTYPE(c_type) _jerry_native_handle_info_##c_type

#define JERRY_DEFINE_NATIVE_HANDLE_INFO(c_type, free_cb) \
  static const jerry_native_handle_info_t JERRY_NATIVE_HANDLE_INFO_FOR_CTYPE(c_type) = { \
  .type = (uintptr_t)&JERRY_NATIVE_HANDLE_INFO_FOR_CTYPE(c_type), \
  .free_cb = free_cb, \
};

// When getting, also pass the native handle info.
// Return false if no native handle OR info doesn't match (latter is new API behavior).
bool jerry_get_object_native_handle_ext (const jerry_value_t obj_val, uintptr_t *out_handle_p, 
                                         const jerry_native_handle_info_t *expected_native_handle_info);

void jerry_set_object_native_handle_ext (const jerry_value_t obj_val, uintptr_t handle_p,
                                         const jerry_native_handle_info_t *native_handle_info);

// --- my_binding.c

typedef struct {
  int x;
  bool y;
} MyBindingThing;

static void my_thing_free_cb(const uintptr_t native_p) {
  // ...
}

JERRY_DEFINE_NATIVE_HANDLE_INFO(MyBindingThing, my_thing_free_cb);

static void my_thing_create(...) {
  MyBindingThing *native_thing = malloc(sizeof(MyBindingThing));
  // ...
  jerry_value_t js_thing = jerry_create_object();
  jerry_set_object_native_handle(js_thing, (uintptr_t)native_thing,
                                 JERRY_NATIVE_HANDLE_INFO_FOR_CTYPE(MyBindingThing));
}

static void get_native_thing(jerry_value_t js_thing) {
  MyBindingThing *native_thing;
  if (!jerry_get_object_native_handle_ext(js_thing, (uintptr_t)&native_thing,     
      JERRY_NATIVE_HANDLE_INFO_FOR_CTYPE(MyBindingThing))) {
    // Either no native handle found *or wrong type* (new API behavior).
    // ... handle the error case
    return;
  }
  // native handle is of expected type, safe to use native_thing now!
  // ...
}

```

jiangzidong, 2017-03-24:
That's a good idea which can support multiple types sharing one free_cp. 

It might be enough to associate native type info with the js object and leave the validation job to binding code. If let `jerry_get_object_native_handle` check the type, it seems can't handle the case where the derived type converts to base type. e.g.
```
// a base struct: widget
struct widget_t {
  char *text;
  char *color;
}

// a derived struct:  button
struct button_t {
  widget_t widget;
  handler onclick;
 ...

void set_text (widget*); //binding js function setText
void set_button_onclick (button_t*, handler); //binding js function setButtonOnclick

==============the js code =========
var button1 = new Button(...) // it will new a button_t object, and stored in button1's native handler
setText(button1, "OK") // the button1's native handler type is button_t, but it can also interpreted as widget_t
```

martijnthe, 2017-03-24:
@jiangzidong: Great point. I do think this is a common case actually.

I'll update the description at the top to contain the latest proposal.

martijnthe, 2017-03-27:
@zherczeg any feedback on the latest proposal?

zherczeg, 2017-03-27:
It sounds ok. However I would not create JERRY_NATIVE_HANDLE_INFO_FOR_CTYPE since some people might want to use enums for types.

martijnthe, 2017-03-27:
> However I would not create JERRY_NATIVE_HANDLE_INFO_FOR_CTYPE since some people might want to use enums for types.

I would like everyone to use the same macros, because it will hopefully "standardize" the way the native handle API will be used, making it easier to reuse, mix & match binding code from different projects.

The advantage of using a `.text` address is that it's like an "automatic" unique ID (only within the context of the finally linked binary of course), whereas an `enum` is centrally defined. That makes mixing & matching code from different projects harder and cumbersome.

Maybe add `#ifndef ...` around those defines? That way they can easily be changed to use an `enum` or otherwise. That said, I think the code space savings are tiny and the benefits I mentioned outweigh the space savings, IMHO.

zherczeg, 2017-03-28:
Actually you have a point. Every use case has different requirements. In same cases a number is enough, in other cases you need Linux kernel like "ops" structures with several callbacks to operations. This could all be done by passing *free_handle to jerry, so everybody could use their own structure as long as the first member is a free_handle. We could also provide a handle/type pair structure as an example for this. This could be relatively easily implemented in Jerry.

martijnthe, 2017-03-28:
I like that idea.
I'll update the proposal in the description again.

One thing I wonder though, if someone chooses to use no other fields aside of the `free_handle`, whether the compiler/linker is allowed to "de-dupe"/merge those const data vars if they contain same data, i.e.:

```
static const jerry_native_handle_info_t info_a = {
  .free_cb = shared_free_cb,
};

static const jerry_native_handle_info_t info_b = {
  .free_cb = shared_free_cb,
};

// Is there the linker allowed to output code where `&info_a == &info_b`?
```

As for the implementation, I think it's probably easiest to create a new magic string literal, also to avoid disrupting the existing native handle API (I'm assuming we want to keep it around and functional for at least a little while.) Agree?

@zherczeg I was wondering why the existing implementation uses 2 "external pointer" properties, instead of having all the info together in one struct (https://github.com/jerryscript-project/jerryscript/blob/master/jerry-core/jerry.c#L1943). It looks like `ecma_create_external_pointer_property()` is only used by `jerry_set_object_native_handle()`. Putting it in one struct would save a bit of memory?

zherczeg, 2017-03-28:
Yes, please create a new magic string literal.

Considering how fast the community is reacted to the external magic string ordering, if adapting the new API requires little work, we could probably just drop the old one. Unless somebody objects.

Yes, you could put everything into one area (rework the code). Probably the motivation was that when only one pointer is defined that uses less memory. I wasn't working on the project that time.

jiangzidong, 2017-03-30:
What about the following interface?
```
bool jerry_get_object_native_handle_ext (const jerry_value_t obj_val,
                                                                  uintptr_t *out_handle_p,
                                                                  uintptr_t *type_info_p);
void jerry_set_object_native_handle_ext (const jerry_value_t obj_val,
                                                                 uintptr_t handle_p,
                                                                 jerry_object_free_callback_t freecb_p,
                                                                 uintptr_t type_info_p);
```
native_handler, free_cp, and type_info will be 3 internal property of object. And developers can put type_info and/or any other related infomations into the type_info_p when they set the handler. When they get it, they also get the saved type_info_p, and can do any check/process on it.

martijnthe, 2017-03-30:
@jiangzidong I'm not in favor of leaving the meaning of the type info value entirely up to the binding developer. Instead, I'd much rather have a design that automatically ensures that binding code from different sources can use the type field in the same way without clashes. The latest proposal (see description at the top) accomplishes that by using `static const jerry_native_handle_info_t my_type_info = { .free_cb = my_free_cb }` for each type. The addresses of the type info variables (i.e. `&my_type_info`) can be used for type comparisons. By using addresses instead of arbitrarily chosen type enum values, they are "automatically" unique** and will thus automatically avoid clashes with other bindings that also use native handles in this way. Note that developers can always "subclass" the `jerry_native_handle_info_t`  struct if they want extra fields. Makes sense?

Re:

> native_handler, free_cp, and type_info will be 3 internal property of object.

I think me / @zherczeg were talking about putting the values together into one struct and having only 1 internal property (to save some memory).

** Assuming you're not using overlays and also not using `-fmerge-all-constants`.

jiangzidong, 2017-03-30:
But if people need to use `-fmerge-all-constants`?

martijnthe, 2017-03-30:
> But what if people need to use -fmerge-all-constants?

Hmm I guess they could remove `const` so instead use `static jerry_native_handle_info_t my_type_info = ...`? I'm not sure if we need to care about that flag, TBH. That flag "breaks" the C language contract:

"Languages like C or C++ require each non-automatic variable to have distinct location, so using this option will result in non-conforming behavior."
https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html

jiangzidong, 2017-03-30:
I will work on it based on this proposal

zherczeg, 2017-03-31:
I don't understand what is the problem if you merge all constants. When you create a const static struct, where the first member is the handler pointer, and two structures has exactly the same data, it shouldn't be a problem. If it is, you can always add an extra type member which prevents merging.

zherczeg, 2017-03-31:
@martijnthe I accidentally updated your comment above. I wanted to reply but somehow the comment was updated. I am sorry, I cannot restore it now.

jiangzidong, 2017-03-31:
@zherczeg I restored it based on my records in mail.

zherczeg, 2017-03-31:
Good idea, I haven't thought of that. Thank you.

jiangzidong, 2017-03-31:
@zherczeg 
>  Actually only two, since the free handler is part of the type info.

It is hard to unify the original `jerry_set_object_native_handle` and new `jerry_set_object_native_handle_ext` in this way. Let me explain it.

`ecma_object` has a property `native_handle_package` whose value is a pointer to a struct. The struct is the `handle_package` which contain the handle, freecb and typeinfo. The struct should be allocated in  `jerry_set_object_native_handle ` and freed in object gc sweep.

If the freecb is not directly a member of the package struct, but a member of typeinfo. For `jerry_set_object_native_handle_ext`, it passes the typeinfo as a argument, so it is easy. While for `jerry_set_object_native_handle`, it passes the freecb, so we need to allocate the typeinfo struct in the function body. But the problem is who will free the "allocated typedinfo struct". We don't know whether the typeinfo struct is passed from external (we shouldn't free during gc) or created by us (we should free). Maybe add another flag in the package struct?

So my suggestion is 3 members in package struct
```
/**
 * Representation for native handle package.
 */

typedef struct
{
  ecma_external_pointer_t handle_p; /**< points to the external native object */
  ecma_external_pointer_t free_cb; /**< free callback for the native handle */
  ecma_external_pointer_t info_p; /**< type info of the native handle */
} ecma_native_handle_package_t;
```
For `jerry_set_object_native_handle`, we just set `info_p= NULL`. What do you think about it?

zherczeg, 2017-03-31:
Shall we keep backward compatibility?

martijnthe, 2017-03-31:
> Shall we keep backward compatibility?

IMO, upgrading to the new API is pretty straightforward.
Personally, when pulling from upstream my expectation is that I may have to change my project slightly.

OTOH, if keeping the old API isn't too much of a cost (runtime speed, memory, ...) maybe keep it around for a little bit (marking it as deprecated)? Aside of deprecating it, maybe it makes sense to re-use the existing names for the new API and add `..._deprecated` to the names of the old functions, to avoid having to carry the weird `..._ext` ending in the future.

Re. @jiangzidong 's question – you could also make the `free_cb` "optional":

```
typedef struct
{
  ecma_external_pointer_t handle_p; /**< points to the external native object */
  ecma_external_pointer_t info_p; /**< type info of the native handle */
  ecma_external_pointer_t free_cb[]; /**< optional (only valid if info_p == NULL) free callback for the native handle */
} ecma_native_handle_package_t;
```

When `info_p` is NULL, `free_cb` is assumed to be valid.
That way you can have a smaller memory footprint when using the new `info_p` and a slightly larger one when using only the old `free_cb`.

jiangzidong, 2017-03-31:
I think so, otherwise all old binding codes will not work. What do you think? @martijnthe

zherczeg, 2017-03-31:
> When info_p is NULL, free_cb is assumed to be valid.

This is a very good idea. Anyway you can create two structs as well. A rough sketch:

    typedef struct
    {
      ecma_external_pointer_t handle_p; /**< points to the external native object */
      ecma_external_pointer_t info_p; /**< type info of the native handle */
    } ecma_native_handle_package_t;

    typedef struct
    {
      ecma_native_handle_package_t base;
      ecma_external_pointer_t free_cb[]; /**< free callback for the native handle */
    } ecma_native_handle_ext_package_t;

I am not sure [] is part of C99 standard.

jiangzidong, 2017-03-31:
@martijnthe Thanks for your reminding.
@zherczeg [] is part of C99, and in your sketch, I think we don't need the [], just 
```
typedef struct
{
  ecma_native_handle_package_t base;
  ecma_external_pointer_t free_cb; /**< free callback for the native handle */
} ecma_native_handle_ext_package_t;
```
Then by checking `if( (ecma_native_handle_package_t *)package->info_p )` we can know which exactly the struct should package cast to.

jiangzidong, 2017-04-01:
Hi, the related PR is #1711

jiangzidong, 2017-04-11:
I will close it since the patch #1711 is already merged.

closed 2017-04-11 by jiangzidong