jerryscript-project/jerryscript #1684: error: zephyr_prebuilt.elf uses VFP register arguments

makerdiary, 2017-03-24:
Hi guys,
I encountered the following problem when compiling jerryscript for zephyr target. I use GNU ARM Embedded Toolchain, not Zephyr SDK on Mac OSX. Any ideas about this?  Thanks for your help.

```
....

../../../../arm-none-eabi/bin/ld: error: zephyr_prebuilt.elf uses VFP register arguments, 
.../../../../arm-none-eabi/lib/libm.a(lib_a-s_nextafter.o) does not

....

```

LaszloLango, 2017-03-24:
Could you share how you try to build? My first guess is that you are trying to use the compilers's libm, but you cannot somehow. Try to use jerry-libm instead.

makerdiary, 2017-03-29:
@LaszloLango Thanks for your quick response.
I just follow the doc to build, but I use  [GNU ARM Embedded Toolchain](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm) instead of Zephyr SDK.

https://github.com/jerryscript-project/jerryscript/blob/master/targets/zephyr/README.md

When I use Zephyr SDK to build, everything is fine.  Maybe it is not incompatible with GCC ARM. I am trying to find out the problem.

closed 2017-04-24 by zherczeg