jerryscript-project/jerryscript #1713: Is there any plan to support for iOS?

yemingyu, 2017-04-04:


zherczeg, 2017-04-04:
Is C code allowed for IOS?

akosthekiss, 2017-04-04:
The code base of jerry-core is pure C99. It already supports and is being continuously tested on/with the arm32 architecture, the clang compiler, and the OSX platform. (And I've also seen jerry running on arm64 on the past.) So, that gives a mix which makes me guess that JerryScript should already compile for iOS without problems (with system libs used instead of jerry-{libc,libm}). That being said, I haven't built jerry for iOS myself, ever.

@yemingyu Are there any obstacles you are facing?

martijnthe, 2017-04-04:
@yemingyu What's the use case for this? Why don't you just use the [JavaScriptCore] library/runtime (it's already included and available for iOS apps).

[JavaScriptCore]: https://developer.apple.com/reference/javascriptcore

yemingyu, 2017-04-05:
@zherczeg Yes

yemingyu, 2017-04-05:
@akosthekiss I want to use a very small engine to excute javaScript，control memory and binary size as small as possible

yemingyu, 2017-04-05:
@martijnthe There are many transformations between JSValue and js object in JavaScriptCore，
this hurts performance

akosthekiss, 2017-04-05:
@yemingyu Have you tried to build JerryScript? If yes, how (what steps have you performed)? What was unsuccessful? (Please give error report/messages if possible.)

zherczeg, 2017-04-05:
I don't see any obstacles porting JerryScript to iOS. If porting the build system is too difficult, you can simply include all .c files (all-in-one build).

akosthekiss, 2017-04-10:
@yemingyu Is there any plan to follow up on this (e.g., by describing unsuccessful build steps and providing the error messages)?

yemingyu, 2017-04-12:
@akosthekiss I am trying to use jerryscript by including all .c files in project like DukTape

akosthekiss, 2017-04-12:
@yemingyu Please, describe your approach in a bit more detail. You might be aware that JerryScript has been supporting all-in-one builds (what you've described above) for a long time. It's there in the cmake-based build system and is also supported by the `tools/build.py` script. Are you using cmake directly or are you building via the python script? Or are you using the sources only and skipping the official build system of JerryScript? And, again, what are the problems you are facing? I cannot find any description of problems or question about any solutions in the discussion.

akosthekiss, 2017-05-05:
Closing due to inactivity (@yemingyu should you still wish to answer the clarification questions, feel free to reopen this issue. Should you have any new questions or run into new problems, please, open a new issue.)

closed 2017-05-05 by akosthekiss

acostin1, 2019-01-30:
Are you allowed by the license to include a different JS engine on iOS than JavaScriptCore?

akosthekiss, 2019-01-30:
@acostin1 If you are asking about the license of JerryScript, it's Apache 2, which is quite permissive and has nothing against iOS or JavaScriptCore. If you are asking about the license of iOS, then this is not the right forum for the question. Such questions should be posted to Apple developer support. (I wouldn't take the responsibility to answer legal questions on Apple's behalf.)