jerryscript-project/jerryscript #1716: [Proposal] Utilities for argument validation in external functions

martijnthe, 2017-04-05:
When implementing external functions, you might have written something like this:

```
// Native implementation of Xyz.prototype.doSomething(a, b)
// First arg is expected to be a string.
// Second (optional) arg is expected to be a boolean.
// `this` is expected to be bound to an instance of Xyz.
static jerry_value_t native_method_impl(const jerry_value_t func_val __attribute__((unused)),
                                        const jerry_value_t this_val __attribute__((unused)),
                                        const jerry_value_t *args_p __attribute__((unused)),
                                        const jerry_length_t args_cnt __attribute__((unused))) {
    if (args_cnt < 2) {
        return jerry_create_error(...);
    }

    // Validate and "transform" (copy to C string) argument 0:
    if (!jerry_value_is_string(args_p[0])) {
        return jerry_create_error(...);
    }
    char arg0[32];
    if (0 == jerry_string_to_utf8_char_buffer(arg[0], (jerry_char_t *)arg0, sizeof(arg0))) {
        // arg0 buffer too small
        return jerry_create_error(...);
    }

    // Validate and "transform" (copy to C bool) argument 1:
    bool arg1 = true; /* default value is true */
    if (args_cnt >= 1) {
        // Optional 2nd arg of type boolean
        if (!jerry_value_is_boolean(args_p[1])) {
            return jerry_create_error(...);
        }
        arg1 = jerry_get_boolean_value(args_p[1]);
    }

    // Validate `this`:
    struct native_obj_t *this_native_obj;
    if (!jerry_get_object_native_handle(this_val, (uintptr_t *)&this_native_obj)) {
        // Whoops, no native handle at all! Caller probably re-bound the method.
        return jerry_create_error(...);
    }
    if (!check_is_native_obj(native_obj)) {
        // Whoops, some other kind of struct! Caller probably re-bound the method.
        return jerry_create_error(...);
    }

    // yay, arguments validated (finally!)
    // Now, let's use this_native_obj, arg0 and arg1 to do something interesting!
    // ... method implementation here ...

    return jerry_create_undefined();
}

void xyz_init(void)
{
    // ...
    jerry_value_t do_something = jerry_create_external_function(native_method_impl);
    jerry_value_t rv = jerry_set_property (xyz_prototype, do_something_name, do_something);
    // ...
}

```

In short, when implementing a native external function, one may have to:
1. Check that mandatory arguments...
  - are supplied at all.
  - are of the expected type.
  - are within the expected range of values.
2. Check optional arguments in a similar manner as mandatory ones. Assume default values when omitted.
3. Check `this` in a similar manner as mandatory ones.
[... 109 lines cut ...]
    jerry_arg_type_t arg_type;

    // "Destination" pointer: (this is where the transformed value need to be assigned to)
    union {
        void *ptr;
        char *ptr_char;
        uint8_t *ptr_uint8;
        // ... etc ...
    };
} jerry_arg_t;

```

Using anonymous unions, defining the array can be quite clean looking.
I imagine adding (inline) C functions to create each of the `jerry_arg_t` elements in the array and get better type safety etc.

`jerry_validate_and_assign_args()` would take the array of `jerry_arg_t` elements and iterate over them, checking for each element whether the corresponding argument from `arg_p` matches. If so, it would transform the value and assign it to the "destination".

## Open Questions
- Extensibility: our current thinking is to add a `JERRY_ARG_CUSTOM` enum value and to provide a validation+transformation callback as part of the `jerry_arg_t` element. Note we would need to add another field for the callback to `jerry_arg_t` (thus increasing all members in the array...) Alternatively, we could have a "registration" API to add custom validations/transforms at runtime. Any other ideas are welcome.
- Errors: would it be OK if the utility defined the Error types and messages? Or do users of this utility need control over this (i.e. through a port function)?
- Smaller binary code size: any thoughts on the current design and its implications for code size? We tried to limit the number of function calls to one (although one with 6 arguments). In our experience, function calls take up quite a lot of code space.
- Alternative solution direction: using a "format string" to specify the transformations in a single C string (kind of like Python's [struct.unpack]). I think this would be harder to use (you have to learn the "format string" format), but it may be better from a code size and extensibility perspective compared to the proposal.

[struct.unpack]: https://docs.python.org/2/library/struct.html

jiangzidong, 2017-04-06:
About the alternative "format string" method: 
* We also need to provide the function to pack/unpack the string, so the code size may not be reduced.
* IMO, it has the same extensibility as the proposal one, because we can always generated the string by the jerry_arg_t array

So I prefer the current proposal.

jiangzidong, 2017-04-07:
What's your opinion about this helper function? @zherczeg @LaszloLango @janjongboom 

I think if Jerry provide such validation APIs/utilities, it will make firmware developers much easier to write binding code. They would be focus on the main logic instead of the tedious part.

I can work on it, and I will add a macro to let users enable/disable the utility.

zherczeg, 2017-04-07:
I agree that such helper functions would be useful. Now we have many proposals and implementations for utility functions, and the key task would be to organize these. The best option would be creating a utility library on top of jerryscript, where multiple variants of such algorithms could be present, and people could select the best option for their use case. Similar to making choices from a menu in a restaurant. E.g. we can have 5-10 argument validation options, and people could select the best one, or make another if they are not satisfied with any of them. This way we could offer choices, and not limit the users freedom. The linker will not add unused functions to the binary anyway. We should move the promise queue implementations there as well. What we need to decide next is creating another project, e.g. jerry-utilities, or keep it in a top level sub-directory inside jerry. Both has pros and cons. What would you prefer? We also need to think about utility dependencies, some utilities might require only a malloc/free, but others might require network connections, etc.

zherczeg, 2017-04-07:
I have been thinking about a priority queue for JerryScript, which would make setTimeout implementation easy, and could be the promise queue as well. This is also just a utility, so it should not be in the core library. But it would be very useful for those, who does not have a natural queue implementation in their system. That would make two queue implementation offered by JerryScript, but it would not be mandatory to use any of them.

martijnthe, 2017-04-07:
> The best option would be creating a utility library on top of jerryscript, where multiple variants of such algorithms could be present, and people could select the best option for their use case. Similar to making choices from a menu in a restaurant.

Totally agree. I'm a big fan of making things modular where possible. Personally I would make separate smaller projects/repos for each of these so that you don't have to drag in one huge monolithic swiss-army knife of a repo.

That said, I think it would be great to have JerryScript-project-"gratified" libraries, especially for foundational things that future libraries may be built on top of and may depend on. If (the interface to) the foundation is too fragmented, I'm afraid we'll never have sane ecosystem of libraries on top.

LaszloLango, 2017-04-10:
I agree. Such utilities must be separated.

janjongboom, 2017-04-10:
@martijnthe  Some things I'm 'missing':

* Automatic memory allocation for strings. I just want them on the heap, so I'd rather have pointer that is set, than allocating the memory first myself (Maybe when you pass in a `const char**`).
* Anything other than simple types... So f.e. returning other objects. I made the [mbed_js_wrap_native_object](https://github.com/janjongboom/mbed-js-http/blob/master/mbed-js-httprequest/mbed-js-httprequest.h#L10) ([used here](https://github.com/janjongboom/mbed-js-http/blob/master/source/mbed-js-httprequest.cpp#L38)) for that, but would be nice to standardize a way of wrapping objects.
* Function pointers.

martijnthe, 2017-04-10:
> Automatic memory allocation for strings. I just want them on the heap, so I'd rather have pointer that is set, than allocating the memory first myself (Maybe when you pass in a const char**).

That's why there's these 2 different ones -- or am I misunderstanding your ask?

```
    // Checks for string, copies C string to existing buffer.
    JERRY_ARG_UTF8_STRING,

    // Checks for string, creates buffer using jerry_port_malloc() and copies C string to it.
    JERRY_ARG_UTF8_STRING_MALLOC,
```

Re. object types: I think if we provide a way to add custom mappings (i.e. `JERRY_ARG_CUSTOM`), that should cover this case, no?

Re. function pointers: can you elaborate a bit?

martijnthe, 2017-04-10:
Re. separate utility library/libraries:

So where to put this particular proposal?

Options mentioned so far:
1. new top level sub-directory inside this repo i.e. `jerry-utils` – only for foundational stuff / boilerplate that 90% of Jerry users need to deal with.
2. separate repo for foundational stuff / boilerplate that 90% of Jerry users need to deal with, i.e. `jerryscript-project/jerryscript-utils`
3. separate repo specifically for the util discussed here i.e. `jerryscript-project/jerryscript-args`

zherczeg, 2017-04-10:
I would rather put all utilities into one project, otherwise it would be too much burden for putting them together (and compile them together), and just use subdirectories for different utility types. This way it would be guaranteed that the whole thing compiles together. So I would prefer both 1 or 2, but if you prefer 3, that is acceptable as well for me.

LaszloLango, 2017-04-10:
I'd prefer 2.

martijnthe, 2017-04-10:
I'd vote for #1, granted the utils are really foundational stuff / boilerplate that 90% of Jerry users need to deal with.

gabrielschulhof, 2017-04-10:
FWIW I also vote 1.

jiangzidong, 2017-04-10:
Both 1 and 2 works for me, and I prefer 1, because I wonder whether it is too early to separate the repo.
In my thought, it's natural to work under jerryscript repo first, if we find there are many candidates/implementations in jerry-utils (and/or many PR/Issues about it), then we create a new repo and move the jerry-utils into it.

zherczeg, 2017-04-10:
I prefer the 1st myself (I proposed the same thing), but lets wait a bit (one day) for others to vote.

janjongboom, 2017-04-10:
@martijnthe Ah, didn't see the MALLOC one. 

Regarding function pointers. Right now you'll have to write custom code to:

* Map from a JS function to a C function, which is hard to script.
* Keep track of the JS function somewhere, so you can clear it when needed.

It'd be nice if that was handled *somewhere*. Not sure if this is the right place.

Regarding the location of these utils I vote for 1. as long as it's properly linked out when not used.

martijnthe, 2017-04-10:
Re. function pointers, can you point me to an example of these?

akosthekiss, 2017-04-10:
Re. separate utility library/libraries:
My vote goes to option 1 (new top level sub-directory inside this repo i.e. `jerry-utils`)

Re. implications for code size:
You've mentioned that you'd add checking+transforming inline functions for each `jerry_arg_type_t` elements (if I get that right, those would be called from `jerry_validate_and_assign_args`, perhaps from within a big switch). Those functions being inline means that they will always be linked into the project that uses the utility function even if some of the arg types are not used. I count 10+ types at least, so this can mean some dead code for a project. (It depends on the size of those functions, of course, and on how big portion of the types are used actually.) If `jerry_arg_t::arg_type` would be a function pointer instead of an enum, and we would have pre-defined (non-inline) check+transform functions for every above-mentioned types (like `JERRY_ARG_BOOL`), then unused check+transform functions would not be kept in the final binary. There would also be no problem with adding custom check+transform functions. (On the downside, indirect function calls are way slower that inlined functions - well, there is no call cost at all.)

Note 1: In the original example, there is a mapping entry `{JERRY_ARG_UTF8_STRING, arg0, sizeof(arg0)}`. How can there be 3 elements, where does that third size field fit in `jerry_arg_t`?

Note 2: I'm not a big fan of untested code. If this goes into the main repo of jerry (I've voted for it with option 1) then we should make sure that not only code from outside tests it. One possibility is to base jerry-main on top of it. Then at least `assert_handler` could use this utility thus ensuring that its core is working.

akosthekiss, 2017-04-10:
Re.(to myself) mapping entry with 3 fields
`jerry_arg_t` could have 3 fields, actually, and the 3rd field could be passed on to check+transform functions as an argument. Its meaning (and size? or not? a union? `void*`) can be function-specific. In the case of the string processing function, this 3rd field could be the size of the available buffer. For the bool processing function, it would be unused (most probably). For a user-defined function, well, whatever the user wants.

jiangzidong, 2017-04-11:
Re jerry_arg_t::arg_type would be a function pointer instead of an enum

We can use macro to prevent binding developers from dealing with the specific check functions. e.g.
    const jerry_arg_t mapping[] = {
        JERRY_ARG_TYPED_NATIVE_HANDLE(&this_native_obj, native_obj_info), 
        JERRY_ARG_UTF8_STRING(arg0, sizeof(arg0)),
        JERRY_ARG_BOOL(&arg1, false), //false means it is optional arg
    };

and in the definition of JERRY_ARG_XXX, we will assign the corresponding check function

akosthekiss, 2017-04-11:
Re jerry_arg_t::arg_type would be a function pointer instead of an enum
I'm not sure I see the benefit of having both functions and macros around them as wrappers. You have to maintain two names then. What is the difference between `JERRY_ARG_TYPED_NATIVE_HANDLE(&this_native_obj, native_obj_info)` and `{jerry_arg_validate_and_assign_native_handle, &this_native_obj, native_obj_info}` (other than that I might have been overly verbose with the function name)?

Re. optionality:
Right now, either a bitfield or a macro parameter (or whatever - none of them see easy to handle/maintain) is needed at each arg to signal optionality. But do we really want to handle/allow optionality on a per-arg basis? Can we have, e.g., `mandatory_arg0, optional_arg1, mandatory_arg2`? (Personally, I don't think so.) Since, if we don't, then `jerry_validate_and_assign_args` could have one more parameter to give the number of mandatory args (which has to be between 0 (inclusive) and `ARRAY_LENGTH(mapping)` (exclusive)) and we could get rid of that bool from the arg descriptions.

jiangzidong, 2017-04-11:
Personally, macro style is like a description and hide the internal details for binding developer, and function style exposes the internal details directly.

> Could have one more parameter to give the number of mandatory args

It works for me.

martijnthe, 2017-04-11:
@akosthekiss sorry, I wasn't clear about the inlined comment:

Re. macros vs functions: I would vote for having `inline` functions to help with the creation of *each* of the struct elements in the array. I think this is better than macros because we'll get type checking of the arguments, code completion, etc. I think it makes sense to have the simple functions that just assign a struct element to be inlined, because an actual function call would be much more expensive just to assign a struct in the array.
**The function that does the actual validation + assignments should NOT be inlined**
So:

```
    const jerry_arg_t mapping[] = {
        // inlined function to create each struct here -- using C function gives us type checking:
        jerry_arg_typed_native_handle(&this_native_obj, native_obj_info), 
        jerry_arg_utf8_string(arg0, sizeof(arg0)),
        jerry_arg_bool(&arg1, false), //false means it is optional arg
    };
    // execute the validation+assignments (NOT an inline function):
    if (!jerry_validate_and_assign_args(&err, this_val, args_p, args_cnt, mapping,  ARRAY_LENGTH(mapping))) {
        return err;
    }
```

> But do we really want to handle/allow optionality on a per-arg basis? Can we have, e.g., mandatory_arg0, optional_arg1, mandatory_arg2? (Personally, I don't think so.)

Yes. There are many JS APIs like this. Trivial example: `Function.prototype.call(thisArg, arg1, arg2, ...)`.
`thisArg` is optional, it can be `undefined` or `null` in case you want it to be the default (the global object).

> Note 2: I'm not a big fan of untested code.
How about having unit tests in `jerry-utils/tests/**` ?

janjongboom, 2017-04-11:
@martijnthe 

[Here's a function](https://github.com/jerryscript-project/jerryscript/blob/master/targets/mbedos5/jerryscript-mbed/jerryscript-mbed-drivers/source/InterruptIn-js.cpp#L28). However, it's really simple, no arguments. But still it's already long (and uses the `wrapFunction` call we added).

For a more complicated one, track [this function](https://github.com/ARMmbed/mbed-js-ble/blob/master/jerryscript-mbed-ble/source/bleCharacteristic.cpp#L74). It's a mess :-)

akosthekiss, 2017-04-11:
Re. inlined functions for struct construction instead of macros
I guess they wont work, C99 doesn't allow that.

```c
typedef struct {
    int x;
} my_s_t;

inline my_s_t my_init_fn(int x) {
    return (my_s_t) { .x = x };
}

const my_s_t arr[] = {
    my_init_fn(1)
};
```

This simplified example, when compiled with `gcc -c -std=c99 const-fn-init.c`, results an error both on Linux (gcc 4.9.3) and on OSX (clang 802.0.41 / LLVM 8.1.0):

```
const-fn-init.c:11:1: error: initializer element is not constant
 };
 ^
const-fn-init.c:11:1: error: (near initialization for ‘arr[0]’)
```
or
```
const-fn-init.c:10:5: error: initializer element is not a compile-time constant
    my_init_fn(1)
```

janjongboom, 2017-04-11:
@akosthekiss Can't you do `return (my_s_t) { x }` ?

akosthekiss, 2017-04-11:
Ah, scratch that. If `arr` is global, it wont work, but if it is within a function, it will. (The example was oversimplified to a point where it didn't replicate the original intent.) BTW, the way the struct initializer is written does not matter, it can be either `{ .x = x }` or `{ x }`.

yichoi, 2017-04-11:
Re. separate utility library/libraries:

I'd prefer 2. Optional features may be in separate repo to keep jerry itself small. If option 1 is accepted, this util would be better by default off. Let the user turn on the feature when they want.

akosthekiss, 2017-04-11:
Re. optionality
@martijnthe given your example, where would the global object come from? I mean, whose responsibility would it be to fill in the reference to global object in arg0 if it was undefined or null? Would the check+transform code do that automatically? Can that be implemented in a general way? Sure, often the global object is the default, but can we deviate from that? I see something similar in the original example, where arg1 is bool and is optional. In the explicit code, it is written out that the default is true if arg1 is unspecified, but in the proposed code there is only either a flag (`JERRY_ARG_OPTIONAL`) or a `false` param (`jerry_arg_bool(&arg1, false), //false means it is optional arg`). How can I specify what I want as the default?

akosthekiss, 2017-04-11:
@yichoi (& all interested parties) My assumption was that since jerry-util was meant to be a subdirectory in the jerryscript repository, it was meant to be a separate library which may or may not be linked to an app that uses JerryScript. The same holds for all(?) jerry-* dirs in the root: jerry-libc, jerry-libm are both optional libs. The only mandatory component is jerry-core. Please, correct me if I'm wrong here. My vote was based on this assumption.

martijnthe, 2017-04-11:
>  In the explicit code, it is written out that the default is true if arg1 is unspecified, but in the proposed code there is only either a flag (JERRY_ARG_OPTIONAL) or a false param (jerry_arg_bool(&arg1, false), //false means it is optional arg). How can I specify what I want as the default?

Ah, sorry, I copy pasted that without reading it carefully. That's not right indeed.
Hmm, I think I'd rather have separate helpers for mandatory vs optional, i.e.:
`jerry_arg_bool(...)` vs `jerry_arg_bool_optional(...)`.

Re. the fallback/default value in case an optional argument is missing / `undefined`:
I was thinking of just assigning the fallback/default to the variable before invoking the check/validation routine (see `bool arg1 = true;` in the original proposal at the top).

We could also add the default value to the helper, i.e. `jerry_arg_bool_optional(&arg1, false /* default */)` but then we need a 4th field in the `jerry_arg_t`. That would put an extra 4 bytes for each element in the array, bloating the whole thing quite a bit...

Should we consider packing the mapping "definition" in some more flexible + less bloat-prone serialized format (perhaps word-aligned to still keep it performant)?

martijnthe, 2017-04-11:
> My assumption was that since jerry-util was meant to be a subdirectory in the jerryscript repository, it was meant to be a separate library which may or may not be linked to an app that uses [...]

That's also my assumption: utils are optional and should not add any code size when not used.

jiangzidong, 2017-04-11:
> I was thinking of just assigning the fallback/default to the variable before invoking the check/validation routine (see bool arg1 = true; in the original proposal at the top).

+1

jiangzidong, 2017-04-12:
I start working on the it and will submit the draft PR soon.

martijnthe, 2017-04-12:
Me and @jiangzidong were just brainstorming on the implementation.
One question arose: do we want to support mapping **multiple** input args in a single validation/transformation function?

For example:
```
// JS Canvas API:
ctx.fillRect(10, 10, 100, 100);

// C structs:
typedef struct { int x; int y; } GPoint;
typedef struct { int w; int h; } GSize;
typedef struct { GPoint origin; GSize size; } GfxRect;
```
(from https://developer.mozilla.org/en-US/docs/Web/API/CanvasRenderingContext2D/fillRect)

In this example, I think it would be ideal to be able to map JS arg0 to arg3 directly to a `GfxRect` with a single `jerry_arg_t` mapping element.

One could also use 4 mapping elements for each arg, but each validation pass would have no knowledge about the other args of the set. In this case it probably does not matter that much, but I can imagine cases where it does matter.

The other thing I like about being able to process multiple input args in one go, is that is would save code size in the mapping array (1 vs 4 is 4x smaller array).

What do you think?

mjessome, 2017-04-12:
> Extensibility: our current thinking is to add a JERRY_ARG_CUSTOM enum value and to provide a validation+transformation callback as part of the jerry_arg_t element. Note we would need to add another [...]

Just to clarify, the extra field in `jerry_arg_t` would be to specify the custom argument type to a single validation+transformation callback?

I would be partial to a lookup table registration API in a similar fashion as the external magic strings API. The addition of a `JERRY_ARG_CUSTOM` flag alongside the `JERRY_ARG_OPTIONAL` could be used to indicate which lookup table to use. This approach feels much cleaner as developers would need to implement their own type => transformation function lookup anyways.

martijnthe, 2017-04-12:
> I would be partial to a lookup table registration API in a similar fashion as the external magic strings API. The addition of a JERRY_ARG_CUSTOM flag alongside the JERRY_ARG_OPTIONAL could be used [...]

Sorry, I haven't had time to update the proposal yet. We're currently thinking of replacing the enum types with function pointers that you'd point to the validator/transformer function. This is more direct and doesn't require a lookup, doesn't require a registration system to support custom validators, etc.

jiangzidong, 2017-04-17:
the first patch is in #1740 . Any review comments are welcome.

akosthekiss, 2017-05-05:
Closing as #1740 has landed (and there was no discussion on expected follow up patches in the short term)

closed 2017-05-05 by akosthekiss