jerryscript-project/jerryscript #1741: snapshot_load_compiled_code does not verify snapshot size

tadziopazur, 2017-04-18:
While jerry_exec_snapshot does receive (a param #2) the length of the snapshot buffer, it does not pass it to snapshot_load_compiled_code, which is prone to unsupported snapshot contents (i.e.: corrupt file, snapshot generated by different jerry version etc.).

This has been observed in practice, crashing TizenRT platform (offset being ~20MB, with largest snapshot included smaller than 5kB).

zherczeg, 2017-04-18:
The version number is checked by Jerry, but no other attempt is made to validate the snapshot. This is intentional, since many environment guarantees a valid snapshot data, and extra checks would be a considerable performance overhead.

Perhaps the simplest solution would be adding some CRC check to your platform. At some point we could think about adding a validating system to jerry utils, but that is a considerable work. You need to validate each byte code operation one by-one, all stack operation one-by-one, all arguments one-by-one. I saw the validating system of action script before, and that is a huge project.

kolipka, 2017-05-17:
Would it be possible to have such mechanism only for debug build? It would be easier to catch some errors. Additionally since snapshot format depends on compile flags, maybe it would be good to verify such things? Simple few ints comparison.

zherczeg, 2017-05-17:
It does not depend on compile flags anymore. That was a bad idea.

What are you trying to do exactly?

tadziopazur, 2017-05-17:
Erm, what do you mean by anymore?
Last I tried I had snapshot generated by host jerry (with 16 bit compressed
pointers, due to jerry heap used).
On arm target, where I used system allocator, the build scripts silently
force 32bit compressed pointers that happen to be stored in a snapshot.
I guess I do not have to tell what the results were..

zherczeg, 2017-05-17:
I thought you were talking about the debugger byte codes. Yes, compressed pointer size affects snapshots. We could add a configuration word besides version to check such features. Is that ok for you?

tadziopazur, 2017-05-17:
TLDR; yes, we need snapshot compatibility validation.

CPTRS are both int the snapshot file, and their length depends on compile
flags.
Proper solution would be to decouple the snapshot format (device
independent would be really great) from in-memory format (this one can be
whatever).

In the lack of such solution we'd like to be able to at least detect such
mismatches when accessing (loading) compiled snapshot.

TIA
-- 
Tomasz Wozniak