jerryscript-project/jerryscript #1746: [Discussion] Runtime configurable heap size

martijnthe, 2017-04-19:
For our project, I would like to be able to instantiate (multiple) JerryScript engines, each with their own heap size.

Currently, the heap size is configured at compile-time using `CONFIG_MEM_HEAP_AREA_SIZE`, which controls these defines and structs:

```
#define JMEM_HEAP_SIZE ((size_t) (CONFIG_MEM_HEAP_AREA_SIZE))
...
#define JMEM_HEAP_AREA_SIZE (JMEM_HEAP_SIZE - JMEM_ALIGNMENT)
...
typedef struct
{
  jmem_heap_free_t first; /**< first node in free region list */
  uint8_t area[JMEM_HEAP_AREA_SIZE]; /**< heap area */
} jmem_heap_t;
...
extern jmem_heap_t jerry_global_heap;
...
// All accesses of `jerry_global_heap` should be done through this macro:
#define JERRY_HEAP_CONTEXT(field) (jerry_global_heap.field)
```

Simplest way to solve this would be to add `#ifndef JERRY_HEAP_CONTEXT` around the `#define JERRY_HEAP_CONTEXT ...` and make it possible to let the client of JerryScript define `JERRY_HEAP_CONTEXT` without having to touch `jcontext.h`. Should the `jerry_global_heap` variable declaration in `jcontext.c` also be #ifdef'd out in this case?

BTW, `JERRY_CONTEXT`, `JERRY_HASH_TABLE_CONTEXT`, etc. should probably also have `#ifndef`s around them?

Thoughts?

martijnthe, 2017-04-19:
cc @jiangzidong

zherczeg, 2017-04-19:
Hm, this is an interesting request. I think the challenge here is that the Jerry allocator does not prepared for variable sized heaps. It expects that the heap size is JMEM_HEAP_AREA_SIZE.

Yes, the definitions of the accessor macros could be improved as well. However they must be defines when jcontext.h is included. So we can add existence checkers around them, but that probably solves only half of the problem. The other side who defines them is still an interesting question. Perhaps jcontext.h should be able to include a custom header which defines these macros.

dbatyai, 2017-04-19:
Actually, the heap does not really depend on `JMEM_HEAP_AREA_SIZE`, as it is only used in a few places. First of all when the heap is initialized, after that all size information is stored within the heap structure, so this shouldn't be a problem. Outside of that, it's also used in the `jmem_is_heap_pointer` helper, and some mem-stat helpers but I think those can also be changed easily.

jiangzidong, 2017-04-20:
My draft plan:
* provide build-time config: JERRY_USE_EXTERNAL_CONTEXT, to indicate the jerry_global_context, jerry_global_heap and jerry_global_hash_table are provided by client (the user of jerry-engine)

* define the jerry_instance_t to represent  a jerry instance
```
typedef struct
{
  void *context_p; //points to the external jerry_context_t
  void *heap_p;//points to the external jmem_heap_t
  void *lcache_p;//points to the external jerry_hash_table_t
  uint32_t heap_size; // like JMEM_HEAP_SIZE
  uint8_t lcache_row_count; // like ECMA_LCACHE_HASH_ROWS_COUNT
  uint8_t lcache_row_length; // like ECMA_LCACHE_HASH_ROW_LENGTH
} jerry_instance_t
```

* in jcontext.c/.h
```
jerry_instance_t *current_instance_p = NULL
#define JERRY_CONTEXT(field) (((jerry_context_t *)current_instance_p->context_p)->field)
#define JERRY_HEAP_CONTEXT(field) (((jmem_heap_t*)current_instance_p->heap_p)->field)

//Also we provide some macro to get the current heap_size, lcache_row_count and lcache_row_length
```

* in client (e.g. main-unix.c)
```
//1. prepare the buffer for context, heap and hashtable
//2. prepare the jerry_instance_t
my_instance = {
...
}
//3. call jerry api to set the current instance
jerry_set_current_instance(&my_instance)

//4 
jerry_init()

... 
```

* Since the jerry_context_t's member will vary with the different build-time configs, and jerry_context_t is not exposed to the jerryscript.h, we should also provide a api to get the sizeof( jerry_context_t).

What do you think about it?

martijnthe, 2017-04-20:
> So we can add existence checkers around them, but that probably solves only half of the problem. The other side who defines them is still an interesting question. Perhaps jcontext.h should be able to include a custom header which defines these macros.

I've used a cflag like `-include my_jcontext.h` in the past to get my own `#define JERRY_CONTEXT` etc. Is this too gross/hacky?

> Perhaps jcontext.h should be able to include a custom header which defines these macros.

That would be a bit cleaner, yeah.

Re. draft: I don't think having one global `current_instance_p` will work well in some situations.
Theoretical example: say you're on any modern OS + multi-core CPU, if you want to spin up multiple threads (each running on a separate core) with each a separate instance this won't work.

How about instead of a setter, having a getter?
So if `JERRY_USE_EXTERNAL_CONTEXT` is defined, use:

```
#define JERRY_CONTEXT(field) jerry_port_get_current_instance()->context_p->(field)
// etc.
```

The port can then define that function (could it still be inlined if that makes sense?).

To create a context, we could also have something like:
```
typedef struct {
  size_t heap_size;
  struct {
   size_t row_count;
   size_t row_length;
  } lcache;
  void (*alloc)(size_t size_in_bytes);
} jerry_instance_config_t;

jerry_instance_t *jerry_instance_create(const jerry_instance_config_t *config);
```
`jerry_instance_t *` can be a fwd declared struct.
This way we're spilling less of the internals.

martijnthe, 2017-04-20:
re. instance getter vs setter, thinking about it a bit more:

I think the problem with `jerry_set_current_instance()` is that the user would need to be able to set it in an OS context switch. I'm not sure if it is very common to have an OS API to get a callback to do something upon a context switch.

On the other hand, usually there is a way to associate some user data with a thread and getting that data when needed: either through a thread local storage API (for example POSIX `pthread_setspecific()` or in Zephyr [k_thread_custom_data_set()]) or C's `__thread` storage specifier. AFAIK, when using these TLS "API"s, it's the responsibility of the OS to handle issues w.r.t. having multiple core under the hood, transparently.

So, having a `jerry_port_get_current_instance()` aligns better with the fact that the underlying system probably already has a way to have a getter for thread-specific data.

[k_thread_custom_data_set()]: https://www.zephyrproject.org/doc/1.7.0/api/kernel_api.html#_CPPv224k_thread_custom_data_setPv

jiangzidong, 2017-04-24:
A draft patch is posted #1778

glistening, 2017-04-24:
Presumably this multiple instance feature implementation will depend on underlying operating system. (Memory allocator and synchronization, thread, ...). What operating system do you think ? Unix and Zephyr ?

martijnthe, 2017-04-25:
> Presumably this multiple instance feature implementation will depend on underlying operating system. (Memory allocator and synchronization, thread, ...). What operating system do you think ? Unix and Zephyr ?

Right. That said, hopefully the design will accommodate most operating systems/RTOSes.
The details of allocating memory should go into the port.

glistening, 2017-04-25:
@martijnthe I hope much part of this implementation can be shared on unix/RTOSes (especially on RTOSes). 

This issue title says "Runtime configurable heap size". However you seem to be thinking of "Support multiple instance of jerry vm". Is it right? If so, there are several things to consider, not only configurable heap size. So, what about making additional issue for other topics?

Before making another issue, I would like to know specify our requirement:

- run multiple instances in the same time? or
- create/terminate vm instance (with resource cleanup) in serial without reboot?

If possible, could you share your use case or scenario about multiple vm instances?

martijnthe, 2017-04-26:
@glistening I've renamed the title of this issue. The discussion started with the need for sizing the heap at runtime, but it evolved into supporting multiple contexts better. Both concerns are taken care of by the same files, `jcontext.h/c`. Both concerns are of importance to our project.

martijnthe, 2017-04-26:
@glistening note that @jiangzidong is working on a patch here to resolve the needs: https://github.com/jerryscript-project/jerryscript/pull/1778

jiangzidong, 2017-05-22:
I will close the issue, because #1778 is merged.

closed 2017-05-22 by jiangzidong