jerryscript-project/jerryscript #1751: Export macro JERRY_SNAPSHOT_VERSION

topelinux, 2017-04-20:
For developer may want to known current snapshot version, could export macro 'JERRY_SNAPSHOT_VERSION' for user to access?
currently this macro exist in jerry-snapshot.h which could not be included by developers.

e.g move JERRY_SNAPSHOT_VERSION from jerry-snapshot.h to jerryscript.h

akosthekiss, 2017-05-05:
@topelinux I'd suggest you open a pull request where you move the macro. That may trigger faster reaction from assigned maintainers.

cc @zherczeg

zherczeg, 2017-05-06:
How can this be useful? I mean you know the current snapshot version is 1234 but what can you use this information for? Statistics?

martijnthe, 2017-05-08:
> How can this be useful? I mean you know the current snapshot version is 1234 but what can you use this information for?

If you are loading snapshots from outside of the system that's running it, this is useful because the system can then ask the snapshot provider for a snapshot that matches the version it is capable of running.

topelinux, 2017-05-08:
The engine that parse/run the snapshot and the tool generated the snapshot from JS files may be two different projects:    
* Project A: The engine run/parse snapshot on MCU(ARM,ESP32...)
* Project B: The tool that generated snapshot which be triggered on PC(X86...)  

Project A and B could have different iteration time, if there is no mechanism to get the SNAP version from outside of Jerryscript, how could  Project A(the engine run/parse snapshot) check if the snap generated by Project B is correct ?

akosthekiss, 2019-01-10:
The macro has been exposed by #2556

closed 2019-01-10 by akosthekiss