jerryscript-project/jerryscript #1752: Interruptible VM loop via callback function

loki04, 2017-04-20:
It would be nice if the VM loop can be interruptible.
In those cases when the JerryScript is used as a library or embedded in a 3rd party application it is hard to be responsible for the application. If the execution of a JavaScript takes too much time - infinite loop, benchmarking, etc. - the main application is not able pause or brake the current execution context.

Is it possible to have a callback function mechanism implemented which is responsible to break or pause the current VM loop? Maybe this callback function can be called in backward loop instruction (if there is such).

martijnthe, 2017-04-20:
@zherczeg brought a similar idea up in the community call yesterday.

I think it makes sense to have this feature.

FWIW, I was planning on running JerryScript in a dedicated RTOS task and adding a software watchdog around it, to make sure that the VM returns control in time. If it doesn't, I'll simply kill the task.

I can imagine scenarios (for example when using cooperative multitasking) where the "interruption" callback is useful.

zherczeg, 2017-04-20:
#1753

loki04, 2017-05-02:
Closing this, since #1753 has been merged.

closed 2017-05-02 by loki04