jerryscript-project/jerryscript #1759: Add FOSA License Scanner to CI

yichoi, 2017-04-21:
As part of foundation ops, we've been looking into ways to add open source license compliance processes to the org in a way that lets JSF projects maintain their autonomy. We're currently looking at adopting FOSSA to help us continuously scan for licenses across JSF projects/their deep dependencies and automate compliance work.  

FOSSA works like a CI or license linter -- on each commit it scans all source files in a project and its dependencies for license violations. It can then automatically trigger Slack notifications, block PRs that bring in deps with incompatible licenses, collect raw copyright headers to generate attribution files/notices and more. They're focused a lot on making compliance really easy to run in the background for developers, and they're already running on a bunch of open source JavaScript projects.

bzsolt, 2017-04-25:
I've checked this tool in order to add into our testing infrastructure, but I've faced some issues which are over my authority. FOSSA requires a lot of permission both in personal and organization/repository area.
E.g.: 
 - This application will be able to read and write all user data.
 - Read and write access to repository webhooks and services.
 - Admin access to organization webhooks.

You can check the requested permissions, when you visit http://fossa.io/ then just click to "Sign up with Github" button.

zherczeg, 2017-05-03:
Perhaps we should create a special user and give these rights to that user?

yichoi, 2017-05-18:
@zherczeg I agree with you

zherczeg, 2017-05-18:
Do you have any name preference? Or shall we choose something?

yichoi, 2017-05-18:
jerry_fossa ?

zherczeg, 2017-05-18:
ok

akosthekiss, 2017-10-11:
It seems that FOSSA support has already been added, at least the readme contains a FOSSA block near the end. However, it must be noted that the continuity of the checks is failing: FOSSA has not been run since mid July. (Moreover, FOSSA is still greedy for privileges, it still requests github login even to see results and R/W access to private data when doing the login with a github account.)

If there is any chance dropping FOSSA, I'd vote for that (because of the outdated and inaccessible results, and its questionable permissions approach).

Alternatively, there is another CI (https://dependencyci.com) which aims at checking licenses of dependencies.
- Pros:
  - easy to set up, can log in with GitHub account, does not ask for write privileges to user data
  - results are accessible without login (e.g., https://dependencyci.com/github/akosthekiss/jerryscript)
- Cons:
  - doesn't seem to comb through all source files of the repository for extra license notices but checks licenses of dependencies only (which requires the use of dependencies via some known package management system, which we don't us except for some python tools).

akosthekiss, 2017-10-11:
One more note: instead of using a CI, we might consider using tools integrated into the Travis workflow.
- FOSSology: huge software, mostly web UI oriented, but has some CLI tools, too. (https://github.com/fossology/fossology, https://www.fossology.org)
- ScanCode toolkit: python-based license scanner (https://github.com/nexB/scancode-toolkit, https://www.nexb.com/open_source_software_compliance.html, https://pypi.python.org/pypi/scancode-toolkit/2.2.1)

rerobika, 2019-10-25:
FOSSA license scanning is available [here](https://app.fossa.io/projects/git%2Bgithub.com%2Fjerryscript-project%2Fjerryscript/refs/branch/master/4bdc3a1c4618e48a4346de0d534e9030a9b2a5ea/preview) so IMO we can finally close this issue.

closed 2019-10-25 by rerobika