jerryscript-project/jerryscript #1763: Assertion 'ECMA_IS_VALUE_ERROR (value)' failed in ecma_get_value_from_error_value

renatahodovan, 2017-04-21:
###### Jerry version:

```
Checked revision: ea44b3b
Build command: ./tools/build.py --clean --debug --profile=es2015-subset
```
###### OS:

```
Ubuntu 16.04.2 LTS
```
###### Test case:

```javascript

new Promise(Math.max.propertyIsEnumerable.toString).then()
```

###### Backtrace:

```
ICE: Assertion 'ECMA_IS_VALUE_ERROR (value)' failed at jerryscript/jerry-core/ecma/base/ecma-helpers-value.c(ecma_get_value_from_error_value):612.
Error: ERR_FAILED_INTERNAL_ASSERTION
bt
#0  0xf7fd8be9 in __kernel_vsyscall ()
#1  0xf7898ea9 in raise () from /lib/i386-linux-gnu/libc.so.6
#2  0xf789a407 in abort () from /lib/i386-linux-gnu/libc.so.6
#3  0x0806c9c3 in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/targets/default/jerry-port-default-fatal.c:53
#4  0x0809c825 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.c:61
#5  0x0809c85d in jerry_assert_fail (assertion=0x8157500 "ECMA_IS_VALUE_ERROR (value)", file=0x8157080 "jerryscript/jerry-core/ecma/base/ecma-helpers-value.c", function=0x8118a00 <__func__.3817.lto_priv.746> "ecma_get_value_from_error_value", line=612) at jerryscript/jerry-core/jrt/jrt-fatals.c:85
#6  0x0810dffe in ecma_get_value_from_error_value (value=4132450179) at jerryscript/jerry-core/ecma/base/ecma-helpers-value.c:612
#7  0x080b69b8 in ecma_process_promise_reaction_job.lto_priv.432 (obj_p=0xf6102170) at jerryscript/jerry-core/ecma/operations/ecma-jobqueue.c:170
#8  0x0806c319 in jerry_port_jobqueue_run () at jerryscript/targets/default/jerry-port-default-jobqueue.c:124
#9  0x0811760f in main (argc=3, argv=0xffffc3d4) at jerryscript/jerry-main/main-unix.c:814
```

<sup>Found by  [Fuzzinator](http://fuzzinator.readthedocs.io/)</sup>

jiangzidong, 2017-04-26:
The bug is fixed in #1729, which is still under review. I will add the regression case in that patch

jiangzidong, 2017-04-26:
fixed in #1729

closed 2017-04-26 by jiangzidong