jerryscript-project/jerryscript #1774: target: tizenrt-artik05x: add minimal working port and the build guide

glistening, 2017-04-24:
I would like to introduce a new port `tizenrt-artik05x`.

Tizen RT is a lightweight RTOS-based platform to support low-end IoT devices. The project is under developing in [github tizenrt repo](https://github.com/Samsung/TizenRT). Artik05x is an IoT SoC solution that will be released soon. Please refer to [Artik053](https://github.com/Samsung/TizenRT/blob/master/build/configs/artik053/README.md) for more information.

I will make a PR that contains the minimum porting that can print `Hello world from jerryscript` and a documen for building and running.

akosthekiss, 2017-05-05:
@glistening #1775 and #1786 have landed some days ago. Do you have plans to submit PRs related to this issue, or has goal been reached and we can close this topic?

glistening, 2017-05-08:
@akosthekiss I will make another PR for tizenrt. I am thinking of checking and getting recent PRs from NuttX port (e.g., setjmp/longjmp, using jerry-libm instead of system libm, and so on). I will follow your suggestion whether I will make it another issue or consider as a part of minimal port.

akosthekiss, 2017-05-08:
@glistening it's completely fine to keep this issue open if you still have related plans.

glistening, 2017-05-17:
Closed since #1822 and #1825 are landed.

closed 2017-05-17 by glistening