jerryscript-project/jerryscript #1779: Merging snapshots to save space

zherczeg, 2017-04-24:
Currently each snapshot has its own string table, and in projects like IoT.js many files have references to the same strings (usually function names). These strings are duplicated for each snapshot file, and the binary size could be reduced by sharing the string table. A multi entry point snapshot file could be a solution for this problem. We need a utility for merging snapshots, and relative offsets would allow merging many files without reaching the maximum snapshot size.

glistening, 2017-05-25:
@zherczeg Thank you for your idea. As you said, IoT.js generates snapshots for each js module files. So it would make sense to merging snapshots. Could you share more in detail ?
- What do you mean _multi entry point_ snapshot file?
- Do you mean we need to use relative offset to avoid reaching the maximum snapshot size?
  If so, do we need to introduce new instruction in byte-code for relative offset?
- Do you have plan to proceed this? If not, maybe I could follow up this item.

glistening, 2017-05-25:
@zherczeg Oh, I just find #1797. I did not look into the PR. I will see the PR first.

zherczeg, 2017-05-25:
Yes, I can share the detail.

- Multi entry: the snapshot generator creates a snapshot which has one primary and multiple subfunctions. If A is the primary, a1, a2.. are the subfunctions, the snapshot looks like this: Aa1a2a3... With this patch we can combine these snapshots into one: Aa1a2a3Bb1b2b3Cc1c2c3 ... To support these we introduce a new snapshot exec function, where you tell the index of the primary function, in this case 0=>A, 1=>B, 2=>C, ...

- Byte code always used relative branches. However functions in the snapshot file used absolute addresses before, e.g. when A references to a1, it was an absolute address. After this patch we use relative addresses (zero offset is the starting offset of the primary function), which is useful when we have a merged snapshot such as Aa1a2a3Bb1b2b3Cc1c2c3 ... 

- Yes. After this patch is landed (it is there for 3 weeks, and I think it is done), we add support for combining snapshots into IoT.js. We will generate snapshots just like now, but after all snapshots are generated, we combine them and create a single snapshot. This single snapshot will be added to IoT.js, and the modules will have indexes rather than byte data. When a built-in module is loaded, we simply get its index, and execute the primary function corresponding to this index.

zherczeg, 2017-05-25:
The gain: the literal pool will be shared, so if multiple snapshots use the same literal (and it is quite frequent), it will be stored only once. The snapshot headers will be eliminated as well, there will be only one snapshot header.

akosthekiss, 2017-10-12:
Closing: #1797 has implemented this feature.

closed 2017-10-12 by akosthekiss