jerryscript-project/jerryscript #1791: Removing jerryscript.h include from jrt.h

zherczeg, 2017-04-28:
It was a shock for me to realize that the public API is available everywhere in the project because it is inluded in jrt.h. JerryScript has an opposite design: we have public/internal aliases for many types. E.g. jerry_value_t (public) is the same as ecma_value_t (private), jerry_length_t (public) = ecma_length_t (private), etc.

However, I realized that now certain types are crawling into the internal side from the public side. This dependency should be dropped. I am willing to work on this.

akosthekiss, 2017-05-04:
I was the one who opposed against something similar in the referenced PR, so I should speak, I guess: OK with me if done consistently.

akosthekiss, 2017-05-06:
While I've been looking at various parts of the code, I found some of the mentioned public types without private counterpart used internally:

- jerry-core/ecma: `jerry_object_free_callback_t`, `jerry_object_native_info_t`, `jerry_external_handler_t`
- jerry-core/jcontext: `jerry_vm_exec_stop_callback_t`

Hope it helps.

BTW, I've run into two strangely named public types: `jerry_user_context_init_cb`, `jerry_user_context_deinit_cb` (they have `_cb` suffix instead of the common `_t`). It's certainly not nice to use multiple naming schemes in the public API. So, if you're already working on the public/private type aliases, I'd suggest to rename these outlier public types, too.

commit 838e74df (2017-05-11): Remove include jerryscript.h from jrt.h. (#1803)

Fixes #1791.

JerryScript-DCO-1.0-Signed-off-by: Zoltan Herczeg zherczeg.u-szeged@partner.samsung.com
closed 2017-05-11 by zherczeg