jerryscript-project/jerryscript #1795: [Discussion] Prefixing "internal global" symbols of jerry-core

akosthekiss, 2017-04-28:
# Problem / Motivation

I've looked around the jerry-core library a bit to see what we expose of ourselves. I was especially interested in the global symbols available from the static `libjerry-core.a` archive. We have two types of global symbols, at least conceptually: there are those which are part of the public API (e.g., `jerry_release_value` -- let's call these "public globals"), and there are those which should probably not be available to the outside world but cannot be source-local `static`s because of modularization (e.g., `vm_run` or `ecma_op_to_string` -- let's call these "internal globals"). We have had problems with "internal globals" before: internal memory management routines used to have a `mem_` prefix but that conflicted with other projects that wanted to embed the library. We ended up renaming the memory management module to `jmem` and re-prefixing all routines with `jmem_`. However, I don't think that the threat of conflicting symbols has been completely lifted.

In the most basic build variant, on OSX, I get the following results when looking for global symbols that _don't_ have the `jerry_` prefix:

```bash
$ ./tools/build.py --clean
$ nm ./build/lib/libjerry-core.a -g | grep -v ' U ' | grep -v 'jerry_' | wc -l
     611
```

This means that there are 600+ global symbols that are only for the internal working of jerry-core but may potentially conflict with code we have no influence on. What's more frightening (to me, at least) is that these "internal globals" follow quite diverse naming schemes:

```bash
$ nm ./build/lib/libjerry-core.a -g | grep -v ' U ' | grep -v 'jerry_' | cut -d ' ' -f 2,3 | cut -d '_' -f 1,2 | sort | uniq -c
   1
   1 ./build/lib/libjerry-core.a(jerry-all-in.c.o):
   2 S _cbc
  35 S _ecma
   2 T _do
 399 T _ecma
  16 T _jmem
  10 T _lexer
  56 T _lit
  15 T _opfunc
  44 T _parser
  19 T _re
   1 T _util
  10 T _vm
```

I think it's quite hard to prepare our users: it's very hard to tell them what identifiers they _must_ avoid if they don't want to run into cryptic linker/runtime(?) issues. (E.g., a `do_`, `util_`, or `re_` prefix is not unimaginable to be used elsewhere.)

_Note_: I'm not concerned about "claiming" the `jerry_` prefix. That should not conflict. If someone chooses this project, s/he should live with the fact that this prefix is reserved for the JS engine.

# Suggestion / Options

**Option 1**: The easy route

Let's rename at least those "internal globals" which do not follow the "prefix with the module name" naming scheme: `do_`, `cbc_`, `util_`. These are the most problematic, I think.

Note: `lexer_`, `parser_`, and `opfunc_` are actually also in violation of that naming scheme, but it's not cast to stone after all. My very subjective judgement does not consider them as harmful as the other three prefixes.

**Option 2**: The middle course

Let's add a newly chosen prefix to _all_ "internal globals". E.g., `jrs_`, `jry_`, `j_`, `j3y_`, `j9t_`, etc. The key is to have a _single_ prefix we can communicate to our users not to use (in addition to `jerry_`).

Note: I haven't listed `jerry_` as a potential prefix for "internal globals". Right now, it's just my instincts that tell me to have different prefixes for public and internal globals.

**Option 3**: The hard path

Let's add a macro that must be used at the definition and at the references of such global symbols, which adds a custom prefix to the symbols. For the biggest flexibility, this macro can have a sensible default for the prefix (see Option 2) but can also allow a config-time override.

```c
#ifndef CONFIG_SYMBOL_PREFIX
#define CONFIG_SYMBOL_PREFIX jrs_
#endif

#define J(SYMBOL) CONFIG_SYMBOL_PREFIX ## SYMBOL

ecma_property_pair_t *
J (ecma_alloc_property_pair) (void)
{
  return J (mem_heap_alloc_block) (sizeof (ecma_property_pair_t));
}
```

Note 1: The example is a trimmed down variant of real code.
Note 2: This option could hopefully also include the reverting of the `mem`-to-`jmem` change, because that single `j` is highly arbitrary IMHO. The example already shows a symbol in reverted form.
Note 3: AFAIK, Valgrind uses a solution something like this.

**Option 4**: Do nothing

We can conclude this, too.

# Analysis

- Option 1:
  - Pro: Can be applied practically at no cost, it's just the renaming of 5 identifiers in the simplest case.
  - Con: Just scratches the surface. In fact, the issue is not solved.
- Option 2:
  - Pro: Issue solved (mostly). Provided that we choose a non-uncommon prefix, we will most probably not conflict with any user.
  - Con: This means a HUGE rewrite of the code base. These functions are everywhere. 600+ definitions and I-cannot-guess-how-many references.
- Option 3:
  - Pro: Issue solved (completely). A user will be able to reconfigure the prefix if that sub-1% chance hits that the default prefix conflicts with its code base.
  - Con: as for Option 3 + with our current coding style ("space before open parenthesis") the rewritten code will look weird (i.e., we cannot even write `J(xxx)`, we have to add that space `J (xxx)`)

Note 1: Should we conclude for any of the above, it will be possible to enforce the naming scheme with checks using similar techniques as shown in Motivation.
Note 2: This should not be a breaking change. Noone should try to access these symbols from outside the library (except for unit tests).

_Any feedback would be greatly appreciated!_

martijnthe, 2017-04-28:
Or run the static library through `objcopy` and make all the internal symbols hidden?

akosthekiss, 2017-04-28:
Hm. Is that possible for archives? They are just a bunch of object files which still have to be linked, aren't they? If you hide the global symbols, the linker will not find the references. Or do I get something wrong? (Do you have a command line example?)

akosthekiss, 2017-04-28:
One additional note: we have use cases (even in `targets`) where jerry-core is used in "source form", not as an archive of compiled binary objects but the source tree itself added to another build system. (Personally, I think that such an approach is suboptimal as we have put quite some effort in writing/tailoring a build system for the needs of jerry -- but I don't know enough about all those other build systems to really judge them.) So, even if we could find a solution for hiding non-API symbols in a static library with `objcopy` (which I could not get done, yet) that would not help these "source form" use cases.

jiangzidong, 2017-04-29:
I will vote for 2 and against 3.

btw, when I use ` nm ./build/lib/libjerry-core.a -g` in linux, it shows no symbols but
```
jerry-snapshot.c.o:
0000000000000001 C __gnu_lto_slim
0000000000000001 C __gnu_lto_v1

jerry.c.o:
0000000000000001 C __gnu_lto_slim
0000000000000001 C __gnu_lto_v1

jerryscript-debugger.c.o:
0000000000000001 C __gnu_lto_slim
0000000000000001 C __gnu_lto_v1

ecma-alloc.c.o:
0000000000000001 C __gnu_lto_slim
0000000000000001 C __gnu_lto_v1
...
```
Do you know why? What should I do to show symbols? (if `nm ./build/bin/jerry`, it will show symbols)

akosthekiss, 2017-04-29:
You have link time optimization enabled. LTO objects cannot be introspected with `nm` AFAIK. They don't contain the usual object code yet but rather still some compiler internal representation. You'll have to turn LTO off (both tools/build.py an CMakeLists have an option for that).

jiangzidong, 2017-04-30:
Yes, it works with `--lto=off`, thanks

martijnthe, 2017-05-01:
> Hm. Is that possible for archives? They are just a bunch of object files which still have to be linked, aren't they? If you hide the global symbols, the linker will not find the references. Or do I get something wrong? (Do you have a command line example?)

Hmm. I did something like that in the past, but that was an iOS project, so maybe specific to Apple's clang.

Are there real-world projects where symbol clashes were actually a problem?

Personally, I think if I would run into a clash problem, I would try to do some post-compilation processing first (on the .o files). I noticed there is `--redefine-syms` option with `objcopy` -- I'd probably use that: https://sourceware.org/binutils/docs/binutils/objcopy.html

Also worth mentioning: in the projects where I used JerryScript so far, I've never used the `.a` that the project's `tools/build.py` produces, but rather added the JerryScript sources myself to the build system of my own project. So I would de-clash it in my own project's build scripts.

martijnthe, 2017-05-01:
> Hmm. I did something like that in the past, but that was an iOS project, so maybe specific to Apple's clang.

Ah, this: http://stackoverflow.com/questions/14259405/pre-link-static-libraries-for-ios-project

akosthekiss, 2017-05-02:
@martijnthe

> never used the .a that the project's tools/build.py produces

Actually, `tools/build.py` is rarely/hardly used by embedders, indeed. I mostly see it used by developers of this project. But the cmake-based build system (which is backing `tools/build.py`) is used by embedders to build the archive (`.a`), both by targets (in-tree) and externally.

> Are there real-world projects where symbol clashes were actually a problem?

Yes, please see #1052 . That was to trigger to rename "mem" to "jmem". (Personally, I think that adding that "j" is just an ugly hotfix to a real problem.)

martijnthe, 2017-05-02:
OK, if people do use the project's CMakeFile directly, maybe just strip the symbols from the .o files as part of the build process?

 ¯\_(ツ)_/¯

akosthekiss, 2017-05-02:
@martijnthe Please, explain. If you strip symbols from the objects of a static archive before linking, you will not be able to link anything together. Stripping symbols after the linking -- if there was a problem, that has happened already. (BTW, does symbol stripping work with link time optimization?)

martijnthe, 2017-05-02:
@akosthekiss Sorry, I meant to write redefine, not strip.

```
~/Desktop$ x86_64-elf-nm helper.o
0000000000000000 T struct_assign
~/Desktop$ x86_64-elf-objcopy --redefine-sym=struct_assign=jerry_struct_assign helper.o
~/Desktop$ x86_64-elf-nm helper.o
0000000000000000 T jerry_struct_assign
```

Note there's a `--redefine-syms` option to pass a text file defining the renames:
https://sourceware.org/binutils/docs/binutils/objcopy.html

I don't think there's a problem linking as long as you process all the .o files.

akosthekiss, 2017-05-02:
@martijnthe For non-LTO objects, that may work. (Ugly, but works.) The question still stands whether it also works with LTO?

martijnthe, 2017-05-02:
... and whether debug symbols are also patched up?

LaszloLango, 2017-05-04:
Sorry for the late answer. I'm not totally against this proposal, but I'm worrying about the blame info. For the suggested solutions the second one is the most preferred by me (with `jry_` or `jrs_` prefix). But in that case whoever will do it, he/she will have a huge code impact. Furthermore it makes harder to track down a change by blame info (I think).
I'm voting for do nothing until this issue got hit by someone like when the `mem` issue happened.

zherczeg, 2017-05-15:
On the long run we should only keep the jerry_ and for practical reasons the ecma_ prefix. I don't want to reserve jerry_ prefix exclusively for public API. But if we go that way, internal stuff should only use the ecma_ prefix.

rerobika, 2019-10-28:
There was no any name collusion for more than 2.5 years also the issue suffers from the lack of the activity. IMO we can close this issue for now please feel free to reopen it anytime when it becomes current again.

closed 2019-10-28 by rerobika