jerryscript-project/jerryscript #1805: Expose dynamic memory management via public API

akosthekiss, 2017-05-08:
It's been an issue several times that users of JerryScript don't have access to dynamic memory management (in the general case). jerry-libc doesn't provide malloc/free, partially because jerry-core doesn't necessarily need them. But that has some adverse effects on use cases:
- jerry-main cannot use dynamic memory for file input reading, or for command line argument handling,
- the default jobqueue port implementation has to access internal jmem functions to allocate memory for job queue items (in its current form -- until #1804 gets merged, if ever), and
- the recent context extension #1802 also needs some dynamic memory to work.

Wouldn't a pair of public API functions (`void *jerry_alloc ()` / `void jerry_free (void *)`) solve these issues? It is of course a somewhat risky thing to allow, because that may easily eat away the jerry heap. But is allowing access to jmem internals any better?

gabrielschulhof, 2017-05-08:
I don't know that the toolchain allows an in-between level that we've talked about before for extensions - that is, it's not JerryScript itself, but it's not a foreign user of JerryScript either, because it's part of the JerryScript project.

Do we want to have this in-between level as a matter of convention? I doubt that would look very different from what we have now, which is reaching into JerryScript internals.

OTOH, having a public API is not bad, because although it may make use of the same heap as JerryScript, ultimately it doesn't matter where you get your memory from if there's not enough of it for what you want to do.

I know for example that Zephyr.js uses `zjs_malloc ()` for dynamic memory, so I guess users could have their own memory management in parallel.

gabrielschulhof, 2017-05-08:
Of course, this (probably incorrectly) reminds me of the segmented memory of the DOS days *shudder*

martijnthe, 2017-05-08:
I like the *idea* of having a single heap for your JS application code (vs having to worry about balancing the sizes of both the Jerry heap a second "native" heap).

Using the Jerry heap for everything that the application code may want to allocate (incl indirect allocation, i.e. through bindings to native C code), has the downside that the large alignment (8 bytes I believe) on the Jerry heap blocks may be pretty wasteful. But then again, having two separated heaps for one and the same end goal (running some app code) can also be very wasteful...

Another thing to keep in mind: native bindings probably already use the `..alloc/free` functions from the project itself. So unless you'd rewrite them to use `jerry_alloc/free`, you'd still have to balance two heap sizes.

Perhaps it **also** make sense to have a set of port functions (`void *jerry_port_alloc(size_t)` and `void jerry_port_free(void *)`. `jerry-ext` code should use the port alloc/free. This way, the user/project can decide what's the best. The default port implementations could just call straight into `jerry_alloc/free` to make things easy to set up.

zherczeg, 2017-05-08:
Just a note: free memory in JerryScript requires the size of the block. This is a bit uncommon type of free but allows saving space.

akosthekiss, 2017-05-08:
I didn't want to advocate the ubiquitous use of the proposed jerry_alloc / jerry_free functions. If an application needs dynamic memory and knows that its platform provides some alloc/free routines, it should/may use those functions. But a generic code, that is based on jerryscript and is not tied to any os/platform/environment (jerryx-ext is an example of that) and needs dynamic memory -- it is now in trouble. I honestly don't think that deep-linking into jmem internals is a nice or proper solution, neither for "really foreign" nor for "not that foreign" code.

About single vs segmented heap: my motivation was not to solve that dilemma. In practice, for those use cases, which use jmem internals for allocating dynamic memory, calling jerry_alloc would not change anything (it would only give cleaner interfacing with the library). The allocation would happen from the same jerry heap. (Irrespectively of the definition of JERRY_SYSTEM_ALLOCATOR, which can force jmem to allocate via system malloc.)

The proposed idea was also not about port functions - at least, not for now. I'm not sure that we can expect dynamic memory handling from all ports. (And one note: not all platforms have weak aliases. I've recently experimented with them, and to my biggest surprise, OSX - or the Mach-O file format - didn't support them. That's certainly something I wouldn't have expected.)

gabrielschulhof, 2017-05-09:
How about implementing the ability to declare your own heap at compile time?

akosthekiss, 2017-05-09:
@gabrielschulhof Please, elaborate. Would that be a direct solution for this issue? Could non-core users rely on the existence of that heap, and what would be the API to access it?

gabrielschulhof, 2017-05-09:
Basically, add an extra heap pointer parameter to the existing `_alloc_block` and `_free_block` functions, and expose those, then implement `jmem_heap_alloc_block ()` and `jmem_heap_free_block ()` as calls to those functions.

Add a macro to define a named heap, and perhaps macros to encapsulate referring to a heap (like the ones in my context PR, in the `jerry-ext/private-headers/section-utils.h`).

This is probably overly simplistic, but that's my basic idea.

akosthekiss, 2017-05-14:
@gabrielschulhof I'm not sure this would solve the problem in the general case. If you don't know for sure that an external heap is defined, then you still cannot use an allocator or you are still bound to call jmem_* internal functions. (Alternatively, if the external heap would not be an option only but would be enforced, then you'd put extra implementation burden on _all_ ports.) On the other hand, the existing jmem_* functions would become slower because of the increased cost of the function calls (+1 param to be passed) - these are very hot functions.

gabrielschulhof, 2017-05-15:
@akosthekiss how much more work is it to dereference a pointer once in alloc() and once in free()? Because I'm thinking:

1. We store a pointer to the current heap in a global static: `jerry_heap_t *jerry_current_heap = &jerry_global_heap;`

2. `jmem_heap_alloc_block()` and `jmem_heap_free_block()` dereference that pointer and do what they do now.

3. We create two public functions which either have an extra `jmem_heap_t *` parameter, or we allow the definition of a public heap at compile time and then they have the same number of parameters as `jmem_heap_alloc_block()`, and they each

```c
jmem_heap_t *previous_heap = jerry_current_heap;
jerry_current_heap = public_heap;
void *result = jmem_heap_alloc_block();
jerry_current_heap = previous_heap;
return result;
```
and similarly for `free()`.

This is probably still simplistic but at least it should only slow down alloc()/free() by an additional pointer dereference.

PR #2503 (2018-09-04): Promote dynamic memory management from debugger transport to core API
Under the cover of the debugger transport layer, allocation on the
engine's heap has been made available to the public. As there are
actually no restrictions on what the allocated memory can be used
for, the memory management functions better fit in the core part
of the API.

Closes #1805

JerryScript-DCO-1.0-Signed-off-by: Akos Kiss akiss@inf.u-szeged.hu
commit 054717fd (2018-09-04): Promote dynamic memory management from debugger transport to core API (#2503)

Under the cover of the debugger transport layer, allocation on the
engine's heap has been made available to the public. As there are
actually no restrictions on what the allocated memory can be used
for, the memory management functions better fit in the core part
of the API.

Closes #1805

JerryScript-DCO-1.0-Signed-off-by: Akos Kiss akiss@inf.u-szeged.hu
closed 2018-09-04 by akosthekiss