jerryscript-project/jerryscript #1823: Add `setjmp` and `longjmp` functions for tizenrt-artik05x target

glistening, 2017-05-15:
## Problem

Currently tizenrt-artik05x does not support `longjmp/setjmp` correctly.
I tried to use `longjmp/setjmp` implementation from nuttx-stm32f4 port.

But it shows following error:

```
System Information:
        Verarm_undefinedinsn: 
Undefined instruction at 0x4130b58
up_assert: Assertion failed at file:armv7-r/arm_undefinedinsn.c line: 107 task: appmain
```

## Looking for Solution

Since artik05x is cortex-r4, not cortex-m4, I looked for `vstm` and `vldm` in [Cortex R4 Technical Manual](http://infocenter.arm.com/help/topic/com.arm.doc.ddi0363e/DDI0363E_cortexr4_r1p3_trm.pdf). And I found `vldm` in this manual, however, there is no `vstm`. So I removed `vstm` from nuttx-stm32f4 `longjmp/setjmp` and it escapes me from above error. ( `vldm` is does _not_ cause the undefined instruction error. )  So I will expand the `vstm` instruction on tizenrt-artik05x and run some tests in order to make sure it is correct.

robertsipka, 2017-05-15:
Just a question about the setjmp/longjmp implementation that may be a different from the source of the NuttX port. Is the hard-float mode enabled? The stm32f4 NuttX port's setjmp implementation expects that the hard-float mode is used and stores s16-s31 vfp registers to buffer, pointed with r0 register, and increases the register on size of stored data. If not, I think removing the vstm instruction is ok, but also needs to remove the vldm instruction.

glistening, 2017-05-15:
@robertsipka tizenrt-artik05x uses `-mfpu=vfpv3`. But I found that artik05x has no floating-point unit and got confirmed by artik05x team. Thus we can (and have to) remove `vstm` and `vldm` as you commented.

I would like to fix my wrong sentence:

> vldm does not cause the undefined instruction error. 

- `vldm` _also_ causes undefined instruction assertion failure.
If I move `vldm` in place of `vstm` (e.g., setjmp), the same assertion failure occurs.
( longjmp seems not be called in simple hello jerry example. It is the reason `vldm` seems ok although it isn't. )

- [Summary of VFP instructions](https://developer.arm.com/docs/dui0473/l/vfp-instructions/summary-of-vfp-instructions) says both of `vldm` and `vstm` should work in `vfpv3`.

glistening, 2017-05-15:
@robertsipka I will check the correctness on tizenrt-artik05x by running tests under tests/jerry/fail/*. Segment faults from the earlier longjmp/setjmp implementation are expected to be eliminated.

closed 2017-05-17 by glistening