jerryscript-project/jerryscript #1836: Macros to automatically call `jerry_release_value`?

martijnthe, 2017-05-19:
Zephyr.js has a nice macro to declare `const jerry_value_t`s. It leverages the `__attribute__ ((__cleanup__(...)))` feature to automatically call `jerry_release_value()` on the value when the variable goes out of scope:

https://github.com/01org/zephyr.js/blob/ecf2fba29de4dafbd1c8e5b6b3ece636ac237213/src/zjs_util.h#L185

Example use:

```
... {
  ZVAL global = jerry_get_global_object();
  ...
  // no need to call jerry_release_value(global);
  // it will be released automatically by the cleanup function.
}
```

The advantages are quite obvious I think. The disadvantage is that it needs to be supported by the compiler. FWIW, GCC and clang support it.

zherczeg, 2017-05-22:
Yes. This is completely optional, so those compilers who does not support it, will not use it.

akosthekiss, 2017-06-17:
Closing as #1874 has landed

closed 2017-06-17 by akosthekiss