jerryscript-project/jerryscript #1845: Context data retrieval is unsafe

gabrielschulhof, 2017-05-22:
Our current design, where a context handler is given during `jerry_init_with_user_context ()` via an init/deinit pair of function pointers and the pointer is later retrieved via `jerry_get_user_context ()` is unsafe, because for two different contexts you can have two different pairs A and B of function pointers passed to `jerry_init_with_user_context ()` but there is nothing to prevent a call to `jerry_get_user_context ()` from the wrong environment - that is, from an environment where the returned pointer is assumed to be a pointer to a structure B when in fact the context was initialized with the init/deinit pair for A.

The answer is to store the init/deinit pair in a global static structure, a pointer to which can serve as a runtime type id for the `void *` stored on the context. Then, if two different pointers are identified by two different such type IDs, requesting one of the pointers by type ID will return the correct one.

martijnthe, 2017-05-22:
With "environment", do you mean different JerryScript instances?

gabrielschulhof, 2017-05-22:
No, I mean the user case for attaching data to a single context. So, if one use case is a module instance list and another use case is some other kind of framework, then, when the module handler calls `jerry_get_user_context ()` it will assume that the resulting `void *` is a pointer to whatever it uses, whereas when the other kind of framework calls `jerry_get_user_context ()` it will assume that the resulting `void *` will be a pointer to it's own state. As things stand, if you accidentally call into the other framework after having done `jerry_init_with_user_context (JERRY_INIT_EMPTY, jerryx_module_init, jerryx_module_deinit)`, things will likely segfault eventually.

gabrielschulhof, 2017-05-22:
Basically, when you wish to retrieve the `void *` you have to either prove that you are the one who created it in the first place, or you have to provide the means to lazily create and return to you a brand-new `void *` (which is what #1833 does).

gabrielschulhof, 2017-05-22:
In both cases you prove it and provide for lazy creation by providing the pointer to the `jerry_context_data_manager_t` as a parameter to `jerry_get_user_context ()` (or `jerry_get_context_data ()` in the #1833 version).

closed 2017-05-24 by zherczeg