jerryscript-project/jerryscript #1852: mbedos5 target build error

fbmrk, 2017-05-25:
There was a commit on December 13 (#1468) - authored by @janjongboom. It has deleted the main.cpp from the targets/mbedos5/source library by saying that 

> main.cpp is auto-generated and should not be included in this repository

But now, if you want to build the mbedos5, there is not an auto-generated main.cpp, but there is an error message:
```[ERROR] ./BUILD/DISCO_F407VG/GCC_ARM/mbed-os/platform/retarget.o: In function `__wrap_main':
retarget.cpp:(.text.__wrap_main+0xa): undefined reference to `main'
collect2: error: ld returned 1 exit status```

janjongboom, 2017-05-25:
@fbmrk Please build using [mbed-js-example](https://github.com/ARMmbed/mbed-js-example) - which generates the main.cpp file for you depending on the modules that are loaded.

janjongboom, 2017-05-25:
Just tested, and it builds fine against DISCO_F407VG 👍

closed 2017-05-25 by janjongboom