jerryscript-project/jerryscript #1854: Overflow issue when linking all-in-one on ESP32

psiphi75, 2017-05-25:
When building the `jerry-all-in.c` and then compiling it using the ESP32 toolchain, the linker gives the following error:

```text
/opt/Espressif/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/simon/Projects/esp32/build/nibbl.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/opt/Espressif/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: region `dram0_0_seg' overflowed by 231096 bytes
collect2: error: ld returned 1 exit status
/opt/Espressif/esp-idf/make/project.mk:322: recipe for target '/home/simon/Projects/esp32/build/nibbl.elf' failed
```

Below are the build settings (these are based on the [esp8266 target](https://github.com/LaszloLango/jerryscript/blob/esp-revival/targets/esp8266/Makefile.esp8266#L66) for JerryScript):
```sh
tools/build.py  --all-in-one=ON                 \
                --jerry-port-default=ON         \
                --mem-stats=OFF                  \
                --mem-heap=20                   \
                --jerry-libc=OFF                \
                --jerry-libm=ON                 \
                --jerry-cmdline=OFF             \
                --lto=OFF
```

Any ideas?

zherczeg, 2017-05-26:
Hm, maybe the binary size is big now. You should disable features like promises or typed arrays if you don't need them. 231K binary size is pretty big.

LaszloLango, 2017-05-26:
The default profile is es5.1, so there are no promise and typed array by default.

zherczeg, 2017-05-26:
Hm 231K is even bigger then. Perhaps some debugging stuff creeping into the build, hard to tell. Is there a statistic on ESP and ARM code size ratio? E.g. how much space is required for 100K ARM code on ESP?

psiphi75, 2017-05-26:
@LaszloLango @zherczeg I have turned off debugging and built the release copy and I still get the same problem.

When I disable the line with `jerry_init (JERRY_INIT_EMPTY);` it builds fine, even though there is other JerryScript code linked.

I have tried the same code on my desktop PC and it compiles and runs fine.

psiphi75, 2017-05-26:
Issue has been resolved.  Changing the `CONFIG_MEM_HEAP_AREA_SIZE` as found in `config.h` to `(100 * 1024)` solves this issue.  

Thanks for your help.

closed 2017-05-26 by psiphi75