jerryscript-project/jerryscript #1859: jerry_has_property () returns true even on a newly created object

gabrielschulhof, 2017-05-27:
The following test fails at `jerry_has_property ()`:
```c
#include "config.h"
#include "jerryscript.h"
#include "test-common.h"

int
main (void)
{
  TEST_INIT ();

  jerry_init (JERRY_INIT_EMPTY);

  jerry_value_t object = jerry_create_object ();
  jerry_value_t prop_name = jerry_create_string_from_utf8 ((jerry_char_t *) "something");

  TEST_ASSERT (!jerry_has_property (object, prop_name));
  TEST_ASSERT (!jerry_has_own_property (object, prop_name));

  jerry_release_value (object);
  jerry_release_value (prop_name);

  jerry_cleanup ();

  return 0;
} /* main */
```

Why would any prototype in `object`'s prototype chain have the property `"something"`?

Either way, I think it'd be good to cover those two APIs with a test.

gabrielschulhof, 2017-05-27:
OK. NM. The documentation is wrong. `jerry_has_property ()` and `jerry_has_own_property ()` return `jerry_value_t` not `bool`.

LaszloLango, 2017-05-28:
Wrong button :) sorry

closed 2017-05-29 by LaszloLango