jerryscript-project/jerryscript #1861: heap-buffer-overflow in lit_utf8_string_hash_combine

renatahodovan, 2017-05-28:
###### Jerry version:

```
Checked revision: b51b682
Build command: ./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer --compile-flag=-fno-common --compile-flag=-g --jerry-libc=off --static-link=off --strip=off --system-allocator=on --linker-flag=-fuse-ld=gold --error-messages=on
```
###### OS:

```
Ubuntu 16.04.2 LTS
```
###### Test case:

```javascript
{  ( $
&& $ != $ instanceof [ $ , /[\￐]\🀬/ ]
? $ : $
```

###### Backtrace:

```
=================================================================
==9711==ERROR: AddressSanitizer: heap-buffer-overflow on address 0xf51009e5 at pc 0x080a056d bp 0xffe77c28 sp 0xffe77c18
READ of size 1 at 0xf51009e5 thread T0
    #0 0x80a056c in lit_utf8_string_hash_combine jerryscript/jerry-core/lit/lit-strings.c:572
    #1 0x80a056c in lit_utf8_string_calc_hash jerryscript/jerry-core/lit/lit-strings.c:590
    #2 0x80fa3b2 in ecma_new_ecma_string_from_utf8_converted_to_cesu8 jerryscript/jerry-core/ecma/base/ecma-helpers-string.c:327
    #3 0x8085c3f in lexer_construct_regexp_object jerryscript/jerry-core/parser/js/js-lexer.c:1833
    #4 0x806200c in parser_parse_unary_expression.lto_priv.177 jerryscript/jerry-core/parser/js/js-parser-expr.c:647
    #5 0x80522c3 in parser_parse_expression jerryscript/jerry-core/parser/js/js-parser-expr.c:1363
    #6 0x80604cf in parser_parse_array_literal jerryscript/jerry-core/parser/js/js-parser-expr.c:237
    #7 0x8061ff7 in parser_parse_unary_expression.lto_priv.177 jerryscript/jerry-core/parser/js/js-parser-expr.c:641
    #8 0x80522c3 in parser_parse_expression jerryscript/jerry-core/parser/js/js-parser-expr.c:1363
    #9 0x807908c in parser_parse_statements jerryscript/jerry-core/parser/js/js-parser-statm.c:1981
    #10 0x8091426 in parser_parse_source jerryscript/jerry-core/parser/js/js-parser.c:2042
    #11 0x809373b in parser_parse_script jerryscript/jerry-core/parser/js/js-parser.c:2496
    #12 0x8116b76 in jerry_parse jerryscript/jerry-core/api/jerry.c:326
    #13 0x8116cc4 in jerry_parse_named_resource jerryscript/jerry-core/api/jerry.c:381
    #14 0x8113c82 in main jerryscript/jerry-main/main-unix.c:682
    #15 0xf7017636 in __libc_start_main (/lib/i386-linux-gnu/libc.so.6+0x18636)
    #16 0x8048e90  (jerryscript/build/bin/jerry+0x8048e90)

0xf51009e5 is located 0 bytes to the right of 21-byte region [0xf51009d0,0xf51009e5)
allocated by thread T0 here:
    #0 0xf7268dee in malloc (/usr/lib32/libasan.so.2+0x96dee)
    #1 0x80ab088 in jmem_heap_alloc_block_internal jerryscript/jerry-core/jmem/jmem-heap.c:326
    #2 0x80ab140 in jmem_heap_gc_and_alloc_block.lto_priv.397 jerryscript/jerry-core/jmem/jmem-heap.c:362
    #3 0x809bf1d in jmem_heap_alloc_block jerryscript/jerry-core/jmem/jmem-heap.c:408
    #4 0x810d2f7 in ecma_alloc_string_buffer jerryscript/jerry-core/ecma/base/ecma-alloc.c:184
    #5 0x80f9ea3 in ecma_new_ecma_string_from_utf8_converted_to_cesu8 jerryscript/jerry-core/ecma/base/ecma-helpers-string.c:278
    #6 0x8085c3f in lexer_construct_regexp_object jerryscript/jerry-core/parser/js/js-lexer.c:1833
    #7 0x806200c in parser_parse_unary_expression.lto_priv.177 jerryscript/jerry-core/parser/js/js-parser-expr.c:647
    #8 0x80522c3 in parser_parse_expression jerryscript/jerry-core/parser/js/js-parser-expr.c:1363
    #9 0x80604cf in parser_parse_array_literal jerryscript/jerry-core/parser/js/js-parser-expr.c:237
    #10 0x8061ff7 in parser_parse_unary_expression.lto_priv.177 jerryscript/jerry-core/parser/js/js-parser-expr.c:641
    #11 0x80522c3 in parser_parse_expression jerryscript/jerry-core/parser/js/js-parser-expr.c:1363
    #12 0x807908c in parser_parse_statements jerryscript/jerry-core/parser/js/js-parser-statm.c:1981
    #13 0x8091426 in parser_parse_source jerryscript/jerry-core/parser/js/js-parser.c:2042
    #14 0x809373b in parser_parse_script jerryscript/jerry-core/parser/js/js-parser.c:2496
    #15 0x8116b76 in jerry_parse jerryscript/jerry-core/api/jerry.c:326
    #16 0x8116cc4 in jerry_parse_named_resource jerryscript/jerry-core/api/jerry.c:381
    #17 0x8113c82 in main jerryscript/jerry-main/main-unix.c:682
    #18 0xf7017636 in __libc_start_main (/lib/i386-linux-gnu/libc.so.6+0x18636)

SUMMARY: AddressSanitizer: heap-buffer-overflow jerryscript/jerry-core/lit/lit-strings.c:572 lit_utf8_string_hash_combine
Shadow bytes around the buggy address:
  0x3ea200e0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x3ea200f0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x3ea20100: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x3ea20110: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x3ea20120: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
=>0x3ea20130: fa fa fa fa fa fa fa fa fa fa 00 00[05]fa fa fa
  0x3ea20140: 00 00 00 fa fa fa 00 00 00 fa fa fa 00 00 00 fa
  0x3ea20150: fa fa 00 00 00 fa fa fa 00 00 00 fa fa fa 00 00
  0x3ea20160: 00 fa fa fa 00 00 00 fa fa fa 00 00 00 fa fa fa
  0x3ea20170: 00 00 00 fa fa fa 00 00 00 00 fa fa 00 00 00 fa
  0x3ea20180: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Heap right redzone:      fb
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack partial redzone:   f4
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
==9711==ABORTING
```

<sup>Found by [Fuzzinator](http://fuzzinator.readthedocs.io/) with [Grammarinator](https://github.com/renatahodovan/grammarinator)</sup>

renatahodovan, 2017-05-28:
[Test case](https://github.com/jerryscript-project/jerryscript/files/1034611/test.txt)

rerobika, 2017-07-11:
IMHO, #1912 fixed this problem.

#### Jerry version:

Checked revision: 9191268
Build command: 
```
./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer --compile-flag=-fno-common --compile-flag=-g --jerry-libc=off --static-link=off --strip=off --system-allocator=on --linker-flag=-fuse-ld=gold --error-messages=on
```

#### OS
`Ubuntu 16.04.2 LTS`

#### Testcase
```
{  ( $
&& $ != $ instanceof [ $ , /[\￐]\🀬/ ]
? $ : $ ) }
```
#### Result
`Script Error: ReferenceError: $ is not defined`

@akosthekiss @zherczeg As this issue seems to be resolved, shall we close it?

closed 2017-07-12 by LaszloLango