jerryscript-project/jerryscript #1864: Different behavior from toFixed

glistening, 2017-05-30:
**Jerry version:**

```
Checked revision: 708b155
Build command: ./tools/build.py
```

**OS:**
```
Ubuntu 14.04.3 LTS (x86_64)
```

**Test:**
```
$ node
> (1.255).toFixed(2)
'1.25'

$ jsc
>>> (1.255).toFixed(2)
1.25

$ jerry
jerry> (1.255).toFixed(2)
1.26
```

Is it a bug that does not meet the [toFix ECMA Specification](https://www.ecma-international.org/ecma-262/6.0/index.html#sec-number.prototype.tofixed)? When I take a quick look in v8 code, dtoa works differently for `toFixed`, do we need to provide similar implementation for `toFixed`? 

FYI, I got similar different behavior in `toPrecision`, too.

LaszloLango, 2017-05-30:
Duplicate of #1367

glistening, 2017-05-30:
I will close it. Sorry for duplication.

closed 2017-05-30 by glistening