jerryscript-project/jerryscript #1867: Heap Buffer Over-read  in skip_space

JeffJerseyCow, 2017-05-30:
**Description**
skip_spaces skips over non-relevant characters within the JavaScript source – such as spaces. If it detects 0xe2 the macro LEXER_NEWLINE_LS_PS_BYTE_23 looks ahead in memory for the bytes 0x80 0xa9; paragraph separator. JerryScript doesn’t check if the look ahead bytes are valid memory and as such a memory over-read occurs.

**Testcase**
0xe2

**Backtrace**
INFO: Seed: 164027096
INFO: Loaded 1 modules (8401 guards): [0x825f60, 0x82e2a4), 
../a.out: Running 1 inputs 1 time(s) each.
Running: skipspaces000
=================================================================
==9691==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x602000000031 at pc 0x00000058f116 bp 0x7ffe1baa83c0 sp 0x7ffe1baa83b8
READ of size 1 at 0x602000000031 thread T0
    #0 0x58f115 in skip_spaces /home/chris/fuzzing/jerryscript-libFuzzer/jerry-core/parser/js/js-lexer.c:218:13
    #1 0x58d156 in lexer_next_token /home/chris/fuzzing/jerryscript-libFuzzer/jerry-core/parser/js/js-lexer.c:976:3
    #2 0x56c195 in parser_parse_source /home/chris/fuzzing/jerryscript-libFuzzer/jerry-core/parser/js/js-parser.c:1925:5
    #3 0x56c195 in parser_parse_script /home/chris/fuzzing/jerryscript-libFuzzer/jerry-core/parser/js/js-parser.c:2381
    #4 0x5158fd in jerry_parse /home/chris/fuzzing/jerryscript-libFuzzer/jerry-core/jerry.c:309:18
    #5 0x5154f7 in LLVMFuzzerTestOneInput /home/chris/fuzzing/jerryscript-libFuzzer/jerry-main/main-unix-minimal.cc:39:2
    #6 0x5ab194 in fuzzer::Fuzzer::ExecuteCallback(unsigned char const*, unsigned long) /home/chris/tools/Fuzzer/FuzzerLoop.cpp:451:13
    #7 0x5ab3c3 in fuzzer::Fuzzer::RunOne(unsigned char const*, unsigned long) /home/chris/tools/Fuzzer/FuzzerLoop.cpp:408:3
    #8 0x5a05e1 in fuzzer::RunOneTest(fuzzer::Fuzzer*, char const*, unsigned long) /home/chris/tools/Fuzzer/FuzzerDriver.cpp:268:6
    #9 0x5a360b in fuzzer::FuzzerDriver(int*, char***, int (*)(unsigned char const*, unsigned long)) /home/chris/tools/Fuzzer/FuzzerDriver.cpp:620:9
    #10 0x5a0350 in main /home/chris/tools/Fuzzer/FuzzerMain.cpp:20:10
    #11 0x7fa2ac9d082f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #12 0x41cae8 in _start (/home/chris/fuzzing/jerryscript-libFuzzer/jerry-main/a.out+0x41cae8)

0x602000000031 is located 0 bytes to the right of 1-byte region [0x602000000030,0x602000000031)
allocated by thread T0 here:
    #0 0x5112f0 in operator new[](unsigned long) /home/chris/tools/llvm/projects/compiler-rt/lib/asan/asan_new_delete.cc:84
    #1 0x5ab0c9 in fuzzer::Fuzzer::ExecuteCallback(unsigned char const*, unsigned long) /home/chris/tools/Fuzzer/FuzzerLoop.cpp:442:23
    #2 0x5ab3c3 in fuzzer::Fuzzer::RunOne(unsigned char const*, unsigned long) /home/chris/tools/Fuzzer/FuzzerLoop.cpp:408:3
    #3 0x5a05e1 in fuzzer::RunOneTest(fuzzer::Fuzzer*, char const*, unsigned long) /home/chris/tools/Fuzzer/FuzzerDriver.cpp:268:6
    #4 0x5a360b in fuzzer::FuzzerDriver(int*, char***, int (*)(unsigned char const*, unsigned long)) /home/chris/tools/Fuzzer/FuzzerDriver.cpp:620:9
    #5 0x5a0350 in main /home/chris/tools/Fuzzer/FuzzerMain.cpp:20:10
    #6 0x7fa2ac9d082f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)

SUMMARY: AddressSanitizer: heap-buffer-overflow /home/chris/fuzzing/jerryscript-libFuzzer/jerry-core/parser/js/js-lexer.c:218:13 in skip_spaces
Shadow bytes around the buggy address:
  0x0c047fff7fb0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c047fff7fc0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c047fff7fd0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c047fff7fe0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c047fff7ff0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x0c047fff8000: fa fa 01 fa fa fa[01]fa fa fa fa fa fa fa fa fa
  0x0c047fff8010: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff8020: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff8030: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff8040: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c047fff8050: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==9691==ABORTING

zherczeg, 2017-05-31:
In JerryScript it is the user responsibility to pass valid UTF8/CESU8 data to the engine. If the user does not trust the input, they can use `jerry_is_valid_utf8_string` and `jerry_is_valid_cesu8_string` to check it. If these functions returns with `false`, the user should trow an error.

Overall this is a performance optimization for trusted input.

LaszloLango, 2017-08-01:
Fixed

closed 2017-08-01 by LaszloLango