jerryscript-project/jerryscript #1870: TypedArray.prototype.set(array, offset)

glistening, 2017-06-01:
Currently, JerryScript supports selected APIs in TypedArray. For example, [TypedArray.prototype.set(array, offset)`](https://www.ecma-international.org/ecma-262/6.0/index.html#sec-%typedarray%.prototype.set-array-offset) is not supported.

```
// h.js
var buffer = new ArrayBuffer(8);
var uint8 = new Uint8Array(buffer);
uint8.set([1,2,3], 3);
print(uint8);  // console.log for node
```

```
$ node h.js
Uint8Array { '0': 0, '1': 0, '2': 0, '3': 1, '4': 2, '5': 3, '6': 0, '7': 0 }

$ jsc h.js
[object Uint8Array]

$ jerry h.js
Script Error: TypeError
```
Is there a plan to implement remaining APIs?

LaszloLango, 2017-06-01:
cc: @jiangzidong

jiangzidong, 2017-06-01:
@glistening They are in my todo list, including #1832, but recently I am busy in other project in my company. I will start working on them next week. 

If anyone who want to implement those API, please go ahead :)

glistening, 2017-06-02:
@jiangzidong Thank you for answer. I will start to implement above one and in case that I would try to implement more APIs, I will tell you first.

glistening, 2017-08-18:
Close since #1958 landed.

closed 2017-08-18 by glistening