jerryscript-project/jerryscript #1873: Assertion 'JERRY_CONTEXT (jmem_heap_allocated_size) == 0' failed in jmem_heap_finalize

renatahodovan, 2017-06-01:
###### Jerry version:

```
Checked revision: a9e7dd7
Build command: ./tools/build.py --clean --debug
```
###### OS:

```
Ubuntu 16.04.2 LTS
```
###### Test case:

```javascript
if (function() {
    while ({a: a,
            get a() {}
           })
        continue;
    })
    var a;
```

###### Backtrace:

```
Script Error: SyntaxError
ICE: Assertion 'JERRY_CONTEXT (jmem_heap_allocated_size) == 0' failed at jerryscript/jerry-core/jmem/jmem-heap.c(jmem_heap_finalize):184.
Error: ERR_FAILED_INTERNAL_ASSERTION
bt
#0  0x000000000046ee82 in syscall_2 () at jerryscript/jerry-libc/target/posix/jerry-asm.S:59
#1  0x0000000000402760 in raise (sig=6) at jerryscript/jerry-libc/target/posix/jerry-libc-target.c:95
#2  0x0000000000402732 in abort () at jerryscript/jerry-libc/target/posix/jerry-libc-target.c:81
#3  0x0000000000403b8e in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-port/default/default-fatal.c:71
#4  0x000000000042c99d in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.c:58
#5  0x000000000042c9ed in jerry_assert_fail (assertion=0x476540 "JERRY_CONTEXT (jmem_heap_allocated_size) == 0", file=0x4764b0 "jerryscript/jerry-core/jmem/jmem-heap.c", function=0x485aa0 <__func__.3082.lto_priv.272> "jmem_heap_finalize", line=184) at jerryscript/jerry-core/jrt/jrt-fatals.c:82
#6  0x000000000042bd9c in jmem_heap_finalize () at jerryscript/jerry-core/jmem/jmem-heap.c:184
#7  0x000000000042c6a9 in jmem_finalize () at jerryscript/jerry-core/jmem/jmem-allocator.c:51
#8  0x000000000046a659 in jerry_cleanup () at jerryscript/jerry-core/api/jerry.c:195
#9  0x000000000046e34c in main (argc=3, argv=0x7fffffffd178) at jerryscript/jerry-main/main-unix.c:787
```

<sup>Found by  [Fuzzinator](http://fuzzinator.readthedocs.io/) with [Grammarinator](https://github.com/renatahodovan/grammarinator)</sup>

zherczeg, 2017-06-28:
Note: object literal format changed in ES6. The syntax error is completely valid in ES5, but the memory leak is not. I would not change it until we support es6 object literals.

akosthekiss, 2017-06-28:
What do you mean by invalid memory leak? The engine tries to parse an input, throws an error, and finally asserts out at cleanup.

zherczeg, 2017-06-28:
This is what I am talking about as well. The leak is the error, not the syntax error, before somebody misunderstand the problem.

PR #1897 (2017-06-30): Stack data should be freed for while statements after the expression …
Stack data should be freed for while statements after the expression is parsed.

Fixes #1873 .

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
commit f359eb2d (2017-06-30): Stack data should be freed for while statements after the expression is parsed (#1897)

Fixes #1873.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
closed 2017-06-30 by LaszloLango