jerryscript-project/jerryscript #1886: arg.h: improve error messages

martijnthe, 2017-06-09:
The messages with `Error` objects that `arg.h` functions throw when the validation failed are very generic and can be made more helpful.

Improvement suggestions of things to include the error message:

- the index of the value that failed to be validated
- the (stringified) value that failed to be validated
- in case of a type error: the actual type of the value that failed the type check

Q:
- Easiest would be to use `sprintf()` => OK to refactor jerry-libc-printf.c to add this? OTOH, `sprintf()`  would drag in quite a bit of code... Alternatively, we could use `strcat()` and probably get quite far.
- This would obviously increase the code size footprint a bit. Is it necessary to have a build-time config `#define` to enable/disable these richer messages? Is it OK to piggy-back on `JERRY_ENABLE_ERROR_MESSAGES`?

akosthekiss, 2017-06-15:
My two cents:
- `sprintf`: I'm not completely against adding `sprintf` to jerry-libc but I'd be interested in the first estimates of "quite a bit of code".
- `JERRY_ENABLE_ERROR_MESSAGES`: for now, I'd say that we can rely on that macro as the cmake build system defines it for jerry-ext if it is defined for jerry-core (IIRC). But note: I'd use the macro the same way as used in jerry-core: if error messages are disabled then no error message is thrown at all (so it's not a switch for rich/simple error messages but for some/no error messages).

rerobika, 2019-11-06:
`JERRY_ERROR_MESSAGES` macro can be used in jerry-ext as well, and the jerry-libc problem is not actual anymore. However the issue suffers from the lack of interest for 2 years, so IMO we can close this issue for now, please reopen it if needed.

closed 2019-11-06 by rerobika