jerryscript-project/jerryscript #1887: Code coverage tooling

martijnthe, 2017-06-09:
I'd like to be able to get code coverage data out of the JerryScript VM.
We're using lcov+gcov to generate reports for the C code coverage of the unit tests of our JerryScript bindings.
Ideally, the JerryScript tooling would emit coverage data files that lcov can pick up and add those into one report together with the C code coverage data. I've no idea how feasible this is... :) I suspect that the debugger hooks are in part what we'd need to accomplish this.

zherczeg, 2017-06-09:
Are you talking about JS coverage? In IoT.js we simply use the Istanbul tool for getting it.

martijnthe, 2017-06-09:
Yes, JS coverage. How are you using Istanbul when the JS code relies on native bindings that are implemented using the JerryScript C API?

zherczeg, 2017-06-09:
On C part gcov is good enough.

martijnthe, 2017-06-09:
Ah, I see you are using `babel-plugin-istanbul` (https://github.com/Samsung/iotjs/blob/24f2fcb46598b6275a05a1fdf46168c0147b2929/tools/measure_coverage.sh)

Are you running the transpiled code (with the line counter code added) directly in JerryScript?

zherczeg, 2017-08-02:
Yes.

martijnthe, 2017-08-02:
OK, thanks, I'll try that.

closed 2017-08-02 by martijnthe