jerryscript-project/jerryscript #1888: Conflict of math.h

Osamu-Nakamura, 2017-06-12:
We are now developing the app using jerryscript on mbed OS 5 with GCC ARM Embedded, but, we are suffering from the issue that **math.h in the compiler package** conflicts with **jerry-libm/include/math.h**.

Originally, when building the program on mbed OS 5, "**-mfpu=vfpv3**" and "**-mfloat-abi=hard**" are specified as compilation option and hardware support for floating point operation should be actually available on our platform. Thus, it seems that we can exclude **jerry-libm** from the build target.

Am I correct? Is there any impact if we exclude **jerry-libm**? If you could give your opinion, please let me know.

Thanks.

zherczeg, 2017-06-12:
The jerry-libm is optional. You can exclude it. However if your math library is not precise enough (I saw examples before) you can get incorrect results in some cases. It is your own risk.

akosthekiss, 2017-06-15:
@Osamu-Nakamura, @zherczeg is right. Theoretically, and hopefully in practice as well, it is completely up to you what libm and libc implementations you use as long as they are standard-conforming. The two libraries in the repository, jerry-libc and jerry-libm are highly trimmed-down versions of the standard and contain only those functions that are used by jerry-core (and jerry-main). This allows us to have a really small tool at the and. But it also means that if you have an application that embeds jerry-core but requires more functions from libc and/or libm, you'll have to disable the jerry-libX variants and use another implementation, most probably the one shipped with your toolchain.

Osamu-Nakamura, 2017-06-16:
@zherczeg, @akosthekiss, Thank you for the clarifycation. I understood. I would enable another implementation in accordance with the spec of my app.

closed 2017-06-16 by Osamu-Nakamura