jerryscript-project/jerryscript #1890: __attribute__ statements made unconditionally

gabrielschulhof, 2017-06-14:
https://github.com/jerryscript-project/jerryscript/blob/master/jerry-core/jrt/jrt.h#L28-L40 has some `__attribute__` statement which are outside of `#ifdef __GNU_C__`, so they may fail on compilers that don't support such hints.

gabrielschulhof, 2017-06-29:
I've tried testing with
```bash
./tools/run-tests.py --unittests --buildoptions=--compile-flag=-U__GNUC__
```
to try and pretend that gcc doesn't support attributes and it actually gets pretty far towards building, but it gets these warnings:
```
/home/nix/iot/jerryscript/jerry-core/ecma/base/ecma-helpers-value.c: In function ‘ecma_copy_value’:
/home/nix/iot/jerryscript/jerry-core/ecma/base/ecma-helpers-value.c:654:1: error: control reaches end of non-void function [-Werror=return-type]
 } /* ecma_copy_value */
 ^
cc1: all warnings being treated as errors
make[2]: *** [jerry-core/CMakeFiles/jerry-core.dir/ecma/base/ecma-helpers-value.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/nix/iot/jerryscript/jerry-core/ecma/base/ecma-helpers-string.c: In function ‘ecma_string_to_number’:
/home/nix/iot/jerryscript/jerry-core/ecma/base/ecma-helpers-string.c:839:1: error: control reaches end of non-void function [-Werror=return-type]
 } /* ecma_string_to_number */
 ^
cc1: all warnings being treated as errors
make[2]: *** [jerry-core/CMakeFiles/jerry-core.dir/ecma/base/ecma-helpers-string.c.o] Error 1
In file included from /home/nix/iot/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-array-prototype.c:39:0:
/home/nix/iot/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-internal-routines-template.inc.h: In function ‘ecma_builtin_array_prototype_dispatch_routine’:
/home/nix/iot/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-internal-routines-template.inc.h:214:1: error: control reaches end of non-void function [-Werror=return-type]
 } /* DISPATCH_ROUTINE_ROUTINE_NAME */
 ^
cc1: all warnings being treated as errors
make[2]: *** [jerry-core/CMakeFiles/jerry-core.dir/ecma/builtin-objects/ecma-builtin-array-prototype.c.o] Error 1
make[1]: *** [jerry-core/CMakeFiles/jerry-core.dir/all] Error 2
make: *** [all] Error 2
```

LaszloLango, 2017-08-01:
@gabrielschulhof, was this issue fixed in #1863 ?

gabrielschulhof, 2017-08-01:
No, this issue is just an observation I made in core while working on #1863.

akosthekiss, 2018-05-18:
As for `__attribute__`s, #2313 should have solved the problem. However, the error log above comes because of uses of `JERRY_UNREACHABLE`.

#2342 may be related but I think that it will not solve the problem completely.

#2340 may also be related and may open/show the way for several options: a) new compiler-specific code with appropriate definition for `JERRY_UNREACHABLE` may be added to jrt.h, b) a non-empty compiler-agnostic default may be defined for `JERRY_UNREACHABLE` (e.g., `abort ()`? but that may be to drastic, especially for an NDEBUG build), or c) `JERRY_UNREACHABLE` may be defined on the command line for the compiler.

akosthekiss, 2019-01-11:
Closing as the original issue was about unconditional uses of `__attribute__`s, which had been solved.

`JERRY_UNREACHABLE` is still a no-op for compilers not covered in jrt.h or jerryscript-compiler.h, which may still lead to "control reaches end of non-void function" warnings/errors. However, let that be a separate issue if it hits someone. (But option c) from my comment above can give a solution for those situations.)

closed 2019-01-11 by akosthekiss