jerryscript-project/jerryscript #1903: Build options for selecting a set of modules from jerry-ext

glistening, 2017-07-03:
Currently nuttx-stm32f4 port uses only handler module from jerry-ext. (#1816)
tizenrt-artik05x port also want to use only handler.
However, it seems there is no cmake and build.py option.
I have no choice but including all jerry-ext modules, or not.

If so, I want to add build interfaces for both of cmake and build.py.

For build.py, what about using comma separated string? ( e.g., jerry-ext=arg,handler )
I don't want to introduce another parameter name. We may use same parameter name ( `jerry-ext`), and provides `arg,handler` as default value in order to respect history and other ports.

For cmake, I'd like to use jerryscript cmakefile convention for selecting a set of values. Could anyone give a suggestion? I can't find example in jerryscript cmakefiles.

LaszloLango, 2017-07-03:
What's the problem with the arg build? Does jerry-ext build fail on artik05x target?

glistening, 2017-07-03:
@LaszloLango 
> What's the problem with the arg build? 

The problem is that we don't want to increase binary size for not-used jerryx modules. I reduced binary size about 1.1KB on NuttX with #1816 by removing arg module that is not used in nuttx-stm32f4 port.

> Does jerry-ext build fail on artik05x target?

Yes. jerry-ext build fails on artik05x target. I might fix the error if I investigate the error. However, my priority at this moment is building small binary by selecting jerry extension modules without changing jerry-ext/CMakelists.txt manually.

glistening, 2017-07-04:
@LaszloLango I fixed jerry-ext build error in #1906.

LaszloLango, 2017-07-04:
> The problem is that we don't want to increase binary size for not-used jerryx modules. I reduced binary size about 1.1KB on NuttX with #1816 by removing arg module that is not used in nuttx-stm32f4 port.

This is odd. The linker should drop the unused functions. AFAIK, the compilation of 'arg' must not increase the binary size if you don't use it.

martijnthe, 2017-07-04:
@glistening could you check the .map file to see what part of the arg module exactly is causing the 1.1KB increase when not using it?

glistening, 2017-07-04:
@LaszloLango, @martijnthe  For nuttx-stm32f4 port, we generate all-in-one C file. Does linker drop the unused function in all-in-one source? To make sure, I will check it again. And for tizenrt-artik05x port, we currently don't use all-in-one source, I think you're right. I will check the binary size since I fixed the build error (#1906). Thank you.

glistening, 2017-07-04:
Here is the result from nuttx-stm32f4 port. (all-in-one)

**jerry-ext/handler/***
```
$ make
...
find ../../../jerryscript/jerry-ext/handler -name "*.c" | sed -r -e 's/(\.\.\/)*(.+)/#include "\2"/g' > jerry_ext_allin.c
CC:  jerry_ext_allin.c
...
$ ls -al nuttx.bin
-rwxrwxr-x 1 brian brian 241404 Jul  5 08:03 nuttx.bin
```

**jerry-ext/***
```
$ make
...
find ../../../jerryscript/jerry-ext -name "*.c" | sed -r -e 's/(\.\.\/)*(.+)/#include "\2"/g' > jerry_ext_allin.c
CC:  jerry_ext_allin.c
...
$ ls -al nuttx.bin 
-rwxrwxr-x 1 brian brian 244384 Jul  5 08:06 nuttx.bin
```

glistening, 2017-07-04:
I think we can close this issues. I was confused with the all-in-one case. We don't need specify explicitly for generating archive fie since the linker will drop the unused object file automatically.

closed 2017-07-05 by glistening