jerryscript-project/jerryscript #1905: The problem of building Jerryscript for NuttX

limjunsung, 2017-07-03:
Hi,

I want to build Jerryscript for NuttX and NUCLEO-F411RE board. 

I followed the instructions exactly in the https://github.com/jerryscript-project/jerryscript/tree/master/targets/nuttx-stm32f4 except for step 3.

When I followed ‘step 3. Configure NuttX’ in the document, I typed

`.configure.sh nucleo-f4x1re/f411-nsh`

instead of 

`.configure.sh stm32f4discovery/usbnsh`

because I use NUCLEO-F411RE board.

I built Jerryscript and flashed successfully, and the nsh prompt appears on the terminal.

Then, I commanded “jerry” on the nsh shell, the shell prints 

```
nsh> jerry
nsh: jerry: command not found 
nsh> ?
help usage:  help [-v] [<cmd>]

  [           cmp         free        mh          sh          usleep      
  ?           dirname     help        mv          sleep       xd          
  basename    dd          hexdump     mw          test        
  break       echo        kill        pwd         time        
  cat         exec        ls          rm          true        
  cd          exit        mb          rmdir       uname       
  cp          false       mkdir       set         unset       

Builtin Apps:
  jerry
```

Why the jerry command is not working on the shell?

dbatyai, 2017-07-03:
Hi!

The most probable thing is that NuttX fails to allocate a stack for the jerry app. You can use the `free` command in nsh to see how much free space you have left. The default stack size for jerry is 16k, so if you have less than that, then that's definitely the problem.

In this case you can do two things, either reduce the stack size, or reduce the heap size for Jerryscript. Both can be done from NuttX menuconfig in the system applications section.

Hopefully this will solve the issue.

limjunsung, 2017-07-04:
@dbatyai Thanks a lot! I reduced the heap size for Jerryscript, It works~.

May I ask you another question? 

I commanded "free" on the nuttx shell

```
nsh> free
             total       used       free    largest
Mem:         31280       6976      24304      24304
nsh> 
```
I wonder why total mem size is 31280 and used mem size is 6976. That is, how those mem sizes are calculated? I used a board, nucleo-f411re,  with 128k SRAM, 512 Flash mem, and stack size of jerryscript is set to 16k, heap size of that is set to 91k.

Thanks in advance.

dbatyai, 2017-07-04:
You're welcome, I'm happy to hear it's working now!

Generally speaking, what happens is that the space required by static variables will be reserved in the SRAM, this includes the Jerry heap, context, and lcache. After that, what remains of the SRAM will be reserved by NuttX to be used as a heap. The `free` command in nsh will be reporting the statistics of this NuttX heap.

So, if we take into account the 91k jerry heap, the 2k lcache, the size of the context, plus other static variables in NuttX, and subtract those from the 128k total SRAM, that will roughly leave us with the ~31k NuttX heap that you are seeing.

While NuttX is running, whenever there's a need for memory, it will try to allocate it from this heap. This includes everything the OS requires to run correctly, like networking buffers, file descriptors etc. This is what the `used`statistic is reporting.

Of course, whenever you want to start an application in NuttX, the app will require a stack. It is NuttX's responsibility to provide this stack for the application, and it will also try to do this from it's heap. This is why we have to make sure the free space there is large enough for the Jerry stack.

I'm hoping this answers your question.

akosthekiss, 2017-07-12:
@limjunsung If you are happy with the answer, please let us know by closing the issue. If your question is still unanswered, don't hesitate to push the issue further.

akosthekiss, 2017-07-27:
Closing because of inactivity and because issue seems to be answered.

closed 2017-07-27 by akosthekiss