jerryscript-project/jerryscript #1913: Test system for mbedos5

fbmrk, 2017-07-10:
@zherczeg @janjongboom 
Is there any chance of creating a test system, which would test jerryscript on mbedos5? Perhaps make a directory into jerryscript/targets/mbedos5, which would contain the test files or would create a completely new repo for that.

zherczeg, 2018-01-29:
I forgot about #2149

akosthekiss, 2019-01-12:
Pinging this issue. Is this abandoned / forgotten / wishful thinking / a feature request?

janjongboom, 2019-01-14:
@zherczeg @akosthekiss Once we have the IoT.js integration I'd love to have an integration with our CI. We have 800 development boards in various locations, and have a testing framework for automated testing (https://www.youtube.com/watch?v=SGHJiI7BUYM). But we'd need to consider the future of JerryScript in Mbed and how we'd release IoT.js first. cc @thegecko

rerobika, 2019-10-25:
Closing the issue due to inactivity, please reopen it anytime when it becomes current again.

closed 2019-10-25 by rerobika